google/OpenSK #269: OpenSK does not work properly on Microsoft Edge

facilitytinfoil, 2021-01-22:
## Expected Behavior
I can register the OpenSK key on my Google account

## Actual Behavior
When I try to register the OpenSK key on my Google account, it gets stuck on the confirmation page
Here are the logs:
```
two-step-verification?rapt=***:1 Unchecked runtime.lastError: Could not establish connection. Receiving end does not exist.
two-step-verification?rapt=***:1 Unchecked runtime.lastError: Could not establish connection. Receiving end does not exist.
two-step-verification?rapt=***:1 Unchecked runtime.lastError: Could not establish connection. Receiving end does not exist.
two-step-verification?rapt=***:1 Unchecked runtime.lastError: Could not establish connection. Receiving end does not exist.
```

## Steps to Reproduce the Problem

1. Flash the OpenSK firmware to nRF52840-Dongle
2. Open https://myaccount.google.com/signinoptions/two-step-verification on Microsoft Edge
3. Click "Add security key", and click "Next"
4. Tap the button on the OpenSK key

## Specifications

- Version: latest stable
- Platform: Windows

kaczmarczyck, 2021-01-22:
Hi, thanks for reporting! May I ask you to help narrow down the problem?

Since your report mentions Microsoft Edge explicitly, have you tried the same process on different browsers?
Can you use OpenSK on [webauthn.io](https://webauthn.io/)? Or use OpenSK generally worked for you previously?

facilitytinfoil, 2021-01-22:
I haven't tried it with other browsers when I'm creating this issue, since one of my friends can use it with Google Chrome, I think it's this issue is only occurring on Microsoft Edge. however, when I downloaded Google Chrome and try it on my own, it also occurs on Google Chrome, which is very strange.
It's working with all sites that will trigger the Chromium popup for confirmation, including GitHub, Microsoft Account, and webauthn.io.

kaczmarczyck, 2021-01-22:
@jmichelp Can you verify with the latest stable version if OpenSK works on Windows for logging into a Google account?

geofli, 2021-02-04:
I checked, the following U2F command returns 0x6F00 after press button.

000103000000400b100bdc57411add74cbd0e9c236f88ffb2e0b0bf194379a2d99ee795e9993b0a54672b222c4cf95e151ed8d4d3c767a6cc349435943794e884f3d023a8229fd0000

No matter on Windows or Mac, no matter to use Google logon. You test https://u2fdemo.appspot.com/ on mac, same error.

kaczmarczyck, 2021-02-04:
Just to double check, since 0x6F00 is an internal internal exception that might hint at a device that is not fully set up. Did you configure a certificate (mandatory for U2F)?

```bash
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
```

geofli, 2021-02-04:
I did not perform this step. Will test later.

geofli, 2021-02-04:
It can work (google account and https://u2fdemo.appspot.com/ ) after provisioning attestation certificate and private key. This should be the reason.
OS: Windows 10 20H2 19042, Edge: Version 88.0.705.56 (Official build) (64-bit) .

kaczmarczyck, 2021-02-04:
Thanks for checking. I'll keep this open for the issue creator. @facilitytinfoil Please let me know if you narrowed the problem in the meantime.

facilitytinfoil, 2021-02-10:
> ```shell
> ./tools/configure.py \
>     --certificate=crypto_data/opensk_cert.pem \
>     --private-key=crypto_data/opensk.key
> ```
How can I use this command? I tried this on normal and bootloader mode and I still unable to register my key on Google accounts.
Here is the logs:
```
info: Private key is valid.
info: Certificate is valid.
0it [00:00, ?it/s]
```

kaczmarczyck, 2021-02-10:
This looks like your device does not enumerate. That explains why it does not work, it is not visible. Have you followed this step?

> On Linux, you may want to avoid the need for `root` privileges to interact with the key. For that purpose we provide a udev rule file that can be installed with the following command:

    ```shell
    sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/ &&
    sudo udevadm control --reload
    ```

Could you check (or paste) the output of `lsusb` (assuming you are on a Linux machine)?

facilitytinfoil, 2021-02-10:
Result of lsusb:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 001 Device 017: ID 1915:521f Nordic Semiconductor ASA OpenSK
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I have done that, and I even rebooted my machine, however it still shows this `0it [00:00, ?it/s]` and do nothing.
but when I try to run this command with sudo, I got the following error:
```
info: Private key is valid.
info: Certificate is valid.
Traceback (most recent call last):
  File "./tools/configure.py", line 206, in <module>
    main(parser.parse_args())
  File "./tools/configure.py", line 135, in main
    for authenticator in tqdm(get_opensk_devices(args.batch)):
  File "./tools/configure.py", line 67, in get_opensk_devices
    if (dev.descriptor["vendor_id"],
TypeError: tuple indices must be integers or slices, not str
```

kaczmarczyck, 2021-02-10:
Since your device worked on your friend's computer, but not your own, I assume the problem is on HID level. Can you check the output of
```shell
ls -l /dev
```

and look out for entries like `hidraw*`? What are read/write permissions, owner and group?

facilitytinfoil, 2021-02-10:
after changing the ownership of `/dev/hidraw0` to my own account, I got the same error just like when I'm run this command with sudo, what should I do next?

kaczmarczyck, 2021-02-10:
You usually don't have to do that, I was just trying to get to the bottom of this. It's hard to guess and fix the problem remotely :) Even more so if your OpenSK works on different computers, meaning the problem is not with our firmware.

facilitytinfoil, 2021-02-10:
I fixed that script and now my opensk is working, I don't know if that's beacuse of the version of my fido package is different than yours.

and here is the patch:

```py3
--- configure.py        1970-01-01 00:00:00 0000
+++ configure.py.new    1970-01-01 00:00:00 0000
@@ -64,8 +64,8 @@
 def get_opensk_devices(batch_mode):
   devices = []
   for dev in hid.CtapHidDevice.list_devices():
-    if (dev.descriptor["vendor_id"],
-        dev.descriptor["product_id"]) == OPENSK_VID_PID:
+    if (dev.descriptor[1],
+        dev.descriptor[2]) == OPENSK_VID_PID:
       if dev.capabilities & hid.CAPABILITY.CBOR:
         if batch_mode:
           devices.append(ctap2.CTAP2(dev))
@@ -140,7 +140,7 @@
     aaguid = uuid.UUID(bytes=authenticator.get_info().aaguid)
     info(("Programming device {} AAGUID {} ({}). "
           "Please touch the device to confirm...").format(
-              authenticator.device.descriptor.get("product_string", "Unknown"),
+              "OpenSK",
               aaguid, authenticator.device))
     try:
       result = authenticator.send_cbor(

```

closed 2021-02-10 by facilitytinfoil

kaczmarczyck, 2021-02-10:
Thanks for the info.
@jmichelp Maybe interesting for you?

jmichelp, 2021-02-10:
Interesting. Thanks for letting us know.
Could you please tell us which version of the fido2 python package you're using (`pip3 list | grep fido2` should do the job) as well as the operating system you're using to run the python configuration tool? That would greatly help identifying the issue and ensure that our tool works for everyone.

jmichelp, 2021-02-10:
Identified the issue.
Newer versions of `fido2` python package moved from `dict` to `NamedTuple` for the descriptor. I'll update the code.

PR #284 (2021-02-11): Fix `config.py` tool according to the new API of fido2 python package
Fixes #269

Since fido2 v0.8.1 which we developed our configuration tool on, the device descriptor
moved from python `dict` to `NamedTuple`. This PR updates to this new API and ensure 
that we're using the correct version of fido2 module.

- [x] Tests pass
- [x] Appropriate changes to README are included in PR