google/OpenSK #276: Is it possible to program Nordic nRF52840 dongle with nRF Connect Programmer?

alishir, 2021-01-29:
According to [nRF documentation](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nc_programmer%2FUG%2Fnrf_connect_programmer%2Fncp_programming_dongle.html) we could program nRF52840 dongle with [nRF Connect Programmer](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nc_programmer%2FUG%2Fnrf_connect_programmer%2Fncp_introduction.html).
Is it possible to flash OpenSK firmware to nRF52840 dongle with nRF Connect Programmer without using any jlink or ocd programmer?

jmichelp, 2021-01-29:
We haven't tested it but this shouldn't be any different from the DFU programming mode that we support.

You can produce a HEX file that should work with nRF Connect Programmer using the following command line:
```sh
$ ./deploy.py --board=nrf52840_dongle_dfu --programmer=none --opensk
```

As an alternative, if you have the Python package `nrfutil` installed on your machine (`pip3 install --user nrfutil`), you can directly compile and program over usb:
```sh
$ ./deploy.py --board=nrf52840_dongle_dfu --programmer=nordicdfu --opensk
```
This compiles a working firmware then prompts you to put the dongle in bootloader mode (just like with nRF Connect, by pressing the reset button to get a pulsing red LED) and then flashes the dongle without external hardware.

Note that this has some limitations though: if you previously flashed a dongle using an external JTAG programmer, the bootloader has been overwritten. You may still recover that by dumping the bootloader from a new dongle using the external JTAG programmer and then flashing it the same way to the old dongle.

alishir, 2021-01-29:
@jmichelp Many thanks for the prompt and helpful reply.

closed 2021-01-29 by alishir