google/OpenSK #277: OpenSK paniced

akakou, 2021-01-31:
My OpenSK device paniced...Could you help me...?

## Expected Behavior

The expected behavior is that OpenSK device don't panic when device boot.

## Actual Behavior

My OpenSK device panicked and J-Link RTT Viewer showed error to me.

##### error

```
00> NRF52 HW INFO: Variant: AAD0, Part: N52840, Package: QI, Ram: K256, Flash: K1024
00> panicked at 'called `Option::unwrap()` on a `None` value', src/ctap/storage.rs:90:20
00> 
00> panicked at 'Process ctap2 had a fault', /home/akakou/.rustup/toolchains/nightly-2020-06-03-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/libcore/macros/mod.rs:16:9
00>   Kernel version release-1.5-1043-gc5b7a4f2c
```

## Steps to Reproduce the Problem

1. Plug in a nRF52840-DK to my PC. (USB port and JTAG port)
1. Clone the OpenSK repository.
1. Run this commands.
```sh
./setup.sh && ./deploy.py --board=nrf52840dk --opensk --panic-console
```
4. Open and connect JLINK RTT Viewer.
5. Push boot button.

## Specifications

- Version: 
  - OpenSK: https://github.com/google/OpenSK/commit/b0c1b73897816edb757a163943f7b36a55f76a19
  - JLINK: V6.94
- Platform:
  - Manjaro

ia0, 2021-01-31:
Hi @akakou,

Thanks for the report. This seems related to the fact that your previous build was before OpenSK 1.0. The persistent storage is incompatible with those older versions. Could you try the following command to flash the firmware?

```
./deploy.py --board=nrf52840dk --opensk --panic-console --clear-storage
```

Note that this would erase the persistent storage, resulting in loss of anything there (similarly to a reset). However, this would be the only time you need to do this. The current storage is meant to be kept compatible as the firmware updates (at least on the `stable` branch).

akakou, 2021-01-31:
It works! Thank you @ia0 !

closed 2021-01-31 by akakou