google/OpenSK #286: Question: ecdsa-sk and multiple keys

alexandru-g, 2021-02-18:
Hello,

I am wondering what I'm doing wrong in regards to authenticating using multiple dongles.

I flashed 2 nrf52840 dongles without the --regen-keys argument. I then generated a ecdsa-sk ssh key using the first dongle plugged in and all was working well. I could authenticate using that key by pressing the dongle button.

But when I plug the other dongle and try to authenticate, I get the following error: 
`sign_and_send_pubkey: signing failed for ECDSA-SK "/home/user/.ssh/id_ecdsa_sk.pub" from agent: agent refused operation`

Did I miss something? Thank you!

kaczmarczyck, 2021-02-18:
`--regen-keys` is for attestation. That's a feature used to proof the model type to the relying party when making credentials. The actual key material used for authentication is generated randomly, and never leaves OpenSK.

If you want "cloned" OpenSK's, you'd need to implement some sort of secret extraction mechanism. Proceed very carefully if you dare to go down that road. It can defeat the purpose of a security key.

Usually, security keys of the same model have the same attestation key. So, to put it differently, if what you suggest is possible, someone buying the some security key model could sign into SSH as you.

Let me know if that helped!

alexandru-g, 2021-02-18:
I understand now, yes.

Then the only way is to create different ssh keys for each OpenSK dongle and add the public keys for each of them on the remote server. Thank you!

alexandru-g, 2021-02-18:
I now have another question:
I have generated 2 sets of ssh keys, one for each dongle, with the `-O resident` option (if it matters). I have added the public key on the remote server but I have an issue. 

Connecting with `ssh user@server` while the first dongle is plugged in works without any issues. Using the same command while the second dongle is plugged in results in an error (agent refused operation). If I specify the identity of the second dongle using `-i key` then it also works correctly. However, this kind-of beats the purpose of using the "resident" option, doesn't it? I wished I can just plug in the dongle and connect to the server, without needing to specify the correct handle.

Am I making any sense?

kaczmarczyck, 2021-02-18:
Seems like a OpenSSH quirk, but sounds interesting to check. Could you give me the full instructions to reproduce so we're on the same page?

alexandru-g, 2021-02-18:
Steps would be:

1. Plug in dongle 1
2. `ssh-keygen -t ecdsa-sk -O resident`. Public and private key are generated and if you `ssh-add -l`, the new ecdsa fingerprint is there
3. `ssh-copy-id -i new-ecdsa-key user@server`
4. `ssh user@server` works, I need to press the dongle button and all is good
5. Replace dongle 1 with 2
6. `ssh-keygen -t ecdsa-sk -O resident`. Public and private key are generated and if you `ssh-add -l`, the new ecdsa fingerprint is there
7. `ssh-copy-id -i second-ecdsa-key user@server`
8. `ssh user@server` doesn't work, it says `agent refused operation`. If I manually specify the identity for the second public key, then it successfully connects. If I remove the first dongle's fingerprint from ssh-add then `ssh user@server` works. It seems that it tries to authenticate using dongle 1's handle using dongle 2 if both keys are added in ssh-agent.

Separate problem:
1. Plug in any dongle
2. `ssh-keygen -t ecdsa-sk -O resident`. Public and private key are generated and if you `ssh-add -l`, the new ecdsa fingerprint is there
3. Remove all keys from ssh-agent using `ssh-add -D`
4. Try to add it back fom the SK using `ssh-add -K`. The PIN for the authenticator is requested, even if you never set one. If you press enter without entering, nothing happens (key is not added in ssh-agent). Any other input is refused as a wrong PIN (which is correct).

alexandru-g, 2021-02-18:
Any idea how one can set a pin on the OpenSK device?

alexandru-g, 2021-02-20:
@kaczmarczyck It is now somehow working correctly for the first scenario. It's correctly iterating through the keys and I can have any OpenSK dongle plugged in when connecting to the host.
The second scenario doesn't work at all, I'm unable to obtain the key pairs with `ssh-add -K`. I've tried with a Yubikey and the behavior is the same, it doesn't add any key.

kaczmarczyck, 2021-02-22:
For setting a PIN, you can use Chrome settings: chrome://settings/securityKeys "Create a PIN".
Or you go to any test website like [https://webauthn.io/](https://webauthn.io/) and in its setting require user verification. Then you follow the setup flow (enter PIN twice in the first window), but cancel before entering the PIN again to not actually create a credential.

Can you document here what you did to fix your first problem, for me and the rest of the world? :)

closed 2022-05-31 by kaczmarczyck