google/OpenSK #289: Cannot get OpenSK to work on nRF52840-DK

s-schoen, 2021-02-28:
Hi,  
i'm currently experiencing a problem when trying to get OpenSK to work on a nRF52840-DK. It seems to install and flash successfully but FIDO2 does not work at all. All 4 LEDs are blinking rapidly.  
I'm not sure what exactly the problem is and would be grateful for some help.

## Expected Behavior
After flashing the OpenSK firmware registering on a FIDO2 compatible site, e.g. webauthn.io is possible.

## Actual Behavior
Setting up the environment and flashing is states as success but FIDO2 example sides do not work (e.g. webauthn.io).
`lsusb` lists the OpenSK ASA device. The 4 LEDs on the DK are blinking rapidly.

## Steps to Reproduce the Problem
Starting from the cloned repository

1. Install udev rule
2. `./setup.sh` with following output
```
Submodule 'third_party/libtock-rs' (https://github.com/tock/libtock-rs) registered for path 'third_party/libtock-rs'
Submodule 'third_party/tock' (https://github.com/tock/tock) registered for path 'third_party/tock'
Cloning into '/home/steffen/Desktop/OpenSK/third_party/libtock-rs'...
Cloning into '/home/steffen/Desktop/OpenSK/third_party/tock'...
Submodule path 'third_party/libtock-rs': checked out '828c19de9292ddbca0e2da6a161c0c38124c5053'
Submodule path 'third_party/tock': checked out 'c5b7a4f2c89a8c067f0f5786788f4037b32329fd'
[-] Copying additional boards to Tock... DONE.
[-] Applying patch "01-persistent-storage.patch"... DONE.
[-] Applying patch "02-usb.patch"... DONE.
[-] Applying patch "03-usb-debugging.patch"... DONE.
[-] Applying patch "04-additional-boards.patch"... DONE.
[-] Applying patch "05-mpu-fix.patch"... DONE.
[-] Applying patch "06-update-uicr.patch"... DONE.
[-] Applying patch "07-firmware-protect.patch"... DONE.
Signature ok
subject=CN = OpenSK CA
Getting Private key
Signature ok
subject=CN = OpenSK Hacker Edition
Getting CA Private Key
info: syncing channel updates for 'nightly-2020-06-10-x86_64-unknown-linux-gnu'

  nightly-2020-06-10-x86_64-unknown-linux-gnu unchanged - rustc 1.46.0-nightly (feb3536eb 2020-06-09)

info: checking for self-updates
Requirement already up-to-date: tockloader==1.5 in /home/steffen/.local/lib/python3.8/site-packages (1.5.0)
Requirement already up-to-date: six in /usr/lib/python3.8/site-packages (1.15.0)
Requirement already up-to-date: intelhex in /home/steffen/.local/lib/python3.8/site-packages (2.3.0)
Requirement already satisfied, skipping upgrade: pyserial>=3.0.1 in /usr/lib/python3.8/site-packages (from tockloader==1.5) (3.4)
Requirement already satisfied, skipping upgrade: colorama>=0.3.7 in /usr/lib/python3.8/site-packages (from tockloader==1.5) (0.4.4)
Requirement already satisfied, skipping upgrade: pytoml>=0.1.11 in /usr/lib/python3.8/site-packages (from tockloader==1.5) (0.1.21)
Requirement already satisfied, skipping upgrade: argcomplete>=1.8.2 in /home/steffen/.local/lib/python3.8/site-packages (from tockloader==1.5) (1.12.2)
Requirement already satisfied, skipping upgrade: crcmod>=1.7 in /home/steffen/.local/lib/python3.8/site-packages (from tockloader==1.5) (1.7)
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
    Updating crates.io index
  Installing elf2tab v0.6.0
   Compiling libc v0.2.86
   Compiling proc-macro2 v1.0.24
   Compiling version_check v0.9.2
   Compiling autocfg v1.0.1
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.60
   Compiling bitflags v1.2.1
   Compiling unicode-segmentation v1.7.1
   Compiling unicode-width v0.1.8
   Compiling cfg-if v1.0.0
   Compiling vec_map v0.8.2
   Compiling ansi_term v0.11.0
   Compiling strsim v0.8.0
   Compiling lazy_static v1.4.0
   Compiling byteorder v0.5.3
   Compiling textwrap v0.11.0
   Compiling elf v0.0.10
[... 96 lines cut ...]
`lsusb`  =>  
```
...
Bus 001 Device 007: ID 1915:521f Nordic Semiconductor ASA OpenSK
...
```

`dmesg` =>  
```
[14452.037475] hid-generic 0003:1915:521F.0007: hiddev1,hidraw3: USB HID v1.10 Device [Nordic Semiconductor ASA OpenSK] on usb-0000:00:14.0-4/input0
[14455.119301] usb 1-4: USB disconnect, device number 6
[14470.879047] usb 1-4: new full-speed USB device number 7 using xhci_hcd
[14471.059331] usb 1-4: New USB device found, idVendor=1915, idProduct=521f, bcdDevice= 0.01
[14471.059336] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[14471.059338] usb 1-4: Product: OpenSK
[14471.059341] usb 1-4: Manufacturer: Nordic Semiconductor ASA
[14471.059342] usb 1-4: SerialNumber: v1.0
```

Trying to register using FIDO2 on webauthn.io (or any other site) does not work, nothing happens regardless of which button is pressed.

## Specifications

- Version: current stable (commit b0c1b73)
- Platform: nRF52840-DK

jmichelp, 2021-02-28:
Hi Steffen,

thanks for reporting. The LEDs blinking right at the end of step 3 seems to indicate a kernel panic. And therefore, the step 4 wasn't doing anything (according to the output you give, it only validated the crypto material but the board wasn't programmed).

Was your board used before? If there's some content on the flash memory, this can prevent the storage from initializing properly. In any doubt, you can try to erase the area of memory we use for the storage and then reflash OpenSK firmware by using the following command:
`./deploy.py --board=nrf52840dk --clear-storage --opensk`

Note that when you inject the private key and the certificate on step 4, it's expected that the LED start blinking as the board is waiting for a button to be pressed to confirm the action. But it should show a distinctive blinking pattern using the 4 LEDs and not all the 4 LEDs blinking simultaneously (which indicates a crash).

s-schoen, 2021-02-28:
Yes, the board was used before. I actually tested OpenSK on it a few months ago using the same steps and it worked fine.

`./deploy.py --board=nrf52840dk --clear-storage --opensk`  
definitely did something. Now no LED is on at all and the output of the `configure.py` script is identical, so still not key material programming. Also there is no more device listed using `lsusb` nor `dmesg`. Seems like there is not application flashed at all now. Tried flashing again without the `--clear-storage` but same result. No LEDs and not detected by system

jmichelp, 2021-02-28:
Ok. So I guess the `--clear-storage` option was indeed the right track. The persistent storage format was upgraded a while ago to ensure long term firmware upgradability.

But now, the fact that the board is not showing on USB is more puzzling as this is the duty of the kernel. And the above command is supposed to flash both the kernel and the application.
You may try to fully erase the flash (`echo -e "connect\nerase\nq" | JLinkExe -device nrf52840 -if swd -speed 1000` should do the job) and flash again Tock and OpenSK, in case there were some spurious data messing with the kernel.
If this still doesn't work, I would suggest to try to flash a debug version (`./deploy.py --board=nrf52840dk --opensk --debug --panic-console`) and use JLink RTT Viewer to see what's happening.

s-schoen, 2021-03-01:
I tried `echo -e "connect\nerase\nq" | JLinkExe -device nrf52840 -if swd -speed 1000` (had to choose a device because `nrf52840` was not recognized) and re-flashed but indifferent result.

I've never used the JLink RTT Viewer before, so maybe i'm doing something wrong but the output does not seem to be helping either.

```
LOG: J-Link RTT Viewer V6.88c: Logging started.
LOG: Connecting to J-Link via USB...
LOG: Device "NRF52840_XXAA" selected.
LOG: InitTarget() start
LOG: InitTarget() end
LOG: Found SW-DP with ID 0x2BA01477
LOG: DPIDR: 0x2BA01477
LOG: Scanning AP map to find all available APs
LOG: AP[2]: Stopped AP scan as end of AP map has been reached
LOG: AP[0]: AHB-AP (IDR: 0x24770011)
LOG: AP[1]: JTAG-AP (IDR: 0x02880000)
LOG: Iterating through AP map to find AHB-AP to use
LOG: AP[0]: Core found
LOG: AP[0]: AHB-AP ROM base: 0xE00FF000
LOG: CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
LOG: Found Cortex-M4 r0p1, Little endian.
LOG: FPUnit: 6 code (BP) slots and 2 literal slots
LOG: CoreSight components:
LOG: ROMTbl[0] @ E00FF000
LOG: ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB00C SCS-M7
LOG: ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 003BB002 DWT
LOG: ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 002BB003 FPB
LOG: ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 003BB001 ITM
LOG: ROMTbl[0][4]: E0040000, CID: B105900D, PID: 000BB9A1 TPIU
LOG: ROMTbl[0][5]: E0041000, CID: B105900D, PID: 000BB925 ETM
LOG: RTT Viewer connected.
```

jmichelp, 2021-08-16:
Any update on this issue @s-schoen ?

Leo1003, 2021-08-24:
I also have this issue on my previous used nRF52840 dongle.

I tried to erase the storage using modified kernel.
The modified one calls the following codes in the reset handler and panic.
```rust
    pub fn erase_all(&self) {
        let regs = &*self.registers;

        // Put the NVMC in erase mode.
        regs.config.write(Configuration::WEN::Een);

        regs.eraseall.write(EraseAll::ERASEALL::ERASE);

        // Make sure that the NVMC is done. The CPU should be blocked while the
        // erase is happening, but it doesn't hurt to check too.
        while !regs.ready.is_set(Ready::READY) {}
    }
```

Then I flashed the normal OpenSK back, it still keeps flashing the LEDs...

s-schoen, 2021-08-24:
> Any update on this issue @s-schoen ?

I actually never found a solution to this problem. I ended up using the Dongle for development.

jmichelp, 2021-08-25:
By erasing all the flash, you will also erase the pre-programmed DFU bootloader, which means you'll be left with a dongle that cannot be programmed over JTAG anymore.
So if you can still flash over USB, it probably means that your additional code wasn't executed.

Normally we now provide a way to nicely erase the storage for dongle users:
`./configure.py --board=nrf52840_dongle_dfu --no-tockos --erase_storage`
This will flash a Tock application that will erase the storage.
Then you can flash back OpenSK:
`./configure.py --board=nrf52840_dongle_dfu --opensk`

Leo1003, 2021-08-25:
> Normally we now provide a way to nicely erase the storage for dongle users:
> `./configure.py --board=nrf52840_dongle_dfu --no-tockos --erase_storage`
> This will flash a Tock application that will erase the storage.
> Then you can flash back OpenSK:
> `./configure.py --board=nrf52840_dongle_dfu --opensk`

I tried the method and it still doesn't work...
Both on `stable` and `develop` branch.

kaczmarczyck, 2021-11-17:
@Leo1003 Sorry for coming back so late to this. But I just noticed that the command had a typo, the `configure` should actually be `deploy`:
```
./deploy.py --board=nrf52840_dongle_dfu --erase_storage
./deploy.py --board=nrf52840_dongle_dfu --opensk
```

First deploy is to erase all, second is to re-install!

The new PR #408 intends to improve the documenation around `erase_storage`. Not sure if this fixes your problem, but maybe it helps you find it?

kaczmarczyck, 2021-11-19:
Please reopen or create a new issue if the new documentation in #414 does not help you fix your problem!

closed 2021-11-19 by kaczmarczyck