google/OpenSK #300: Proposal: Support  and extend UF2(USB Flashing Format) bootloader, burning firmware more easily and safely

snowyu, 2021-04-07:
* Upgrade firmware: Users just need to drag and drop the firmware onto the `USB virtual disk`.
  * Note: the firmware must be signed with a special private subkey(OpenPGP Key) to allow burning.

Extend the bootloader features:

* store/update the publc keyring for burning.
  * subkey to verify whether burning is allowed
  * main key to verify whetherrevoking/re-creating subkey is allowed

Even more:

* generate and store a private key-ring into the `OpenSK` device safely.
  * show the device's public keys of to the `USB virtual disk`
  * through the file IO to encrypt/sign the data with the device's key.

kaczmarczyck, 2021-04-07:
Thanks for your request!

For reference: [This is UF2](https://github.com/microsoft/uf2), not to be confused with U2F ;)
Generally, [makerdiary](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/opensk/building/) had UF2 working before, so it should be possible.

We do think that firmware upgrades should be easier, and secured by signatures.

axmmisaka, 2021-04-28:
For UF2 (on makerdiary at least), as long as an accompanying bootloader is present, the UF2 file can be easily generated with this script: https://github.com/microsoft/uf2/blob/master/utils/uf2conv.py

The drag-and-drop functionality is mainly limited because flashing the binaries will destroy UICR. This should get resolved after this PR imho: https://github.com/google/OpenSK/pull/186

kaczmarczyck, 2021-11-19:
Hi @snowyu , the DFU mode is soon a first class citizen in our documentation.

I'm not sure I understand the  rest of the message concerning  OpenPGP. Can you explain more or open a new issue for that?

closed 2021-11-19 by kaczmarczyck

snowyu, 2021-11-21:
Ok,  for example, security upgrade a new firmware workflow:

1.  sign the new firmware with your private `opensk-cert.key`(which signed the `opensk.key` and `opensk.key` should confirmed it too) first.
2. enter the flashing mode and drag the new firmware with the signature to the "`USB virtual disk`".
3. the flashing mode will verify the new firmware with the signature, it will upgrade if the signature ok.

In Common node:

1. the "`USB virtual disk`" will show the public key `opensk.key`  file content.
2. the "`USB virtual disk`" will list all registered FIDO public key files.
3. Drag in an operation signature file to list all FIDO private key files and delete private key files

The OpenPGP format was chosen because it is more popular and it is more convenient to import and export. If it is troublesome, use the default format adopted by OpenSK first

kaczmarczyck, 2021-11-22:
I'm not sure I follow the idea. The certificate and `opensk-cert.key` are used for attestation, it does not sign the firmware.

We are currently exploring a new firmware upgrade mechanism that indeed requires signed firmware, see #404. That upgrade will not use DFU though, but run through CTAP.

snowyu, 2021-11-23:
@kaczmarczyck. Sorry, I am not familiar with the development of OpenSK.

I will use the [subkeys concept of OpenPGP](https://wiki.debian.org/Subkeys) to explain more.

Subkeys: consists of multiple keys which're bound to (signed by) a primary (master) key pair.

The `master key` is used for signing other keys, creating subkeys and revoking subkeys. It should always be stored offline.

Then creates two subkeys:

The first subkey is used as a signature key for various operations: list private FIDO keys, delete FIDO keys, update new firmware.  The `public key` of the master key and this subkey will be placed in the hardware(cannot be reset or replaced) for signature verification.

The second subkey is used as the device key (the opensk key) to manage(sign) the FIDO keys etc. This secret private key will be placed in the secure storage area of the hardware.

kaczmarczyck, 2021-11-23:
I think this would conflate different concepts to a degree I'm not comfortable with. The signing keys in OpenSK are different in lifetime requirements:

- The per-account keys (is this what you mean with FIDO keys?) reside inside the storage or are derived from the master secret, that is also stored in persistent storage. Those are generated on demand and differ per user.
- The attestation key is a private key that can be configured into OpenSK and is either shared between lots of devices, or omitted. This is a choice every vendor (or hacker, if you build OpenSK yourself) has to make.
- The upgrade key is used in our new, experimental method to upgrade OpenSK through the app. The upgrades are signed, and are supposed to be the only way to change the firmware in a locked down device in the future. We currently don't offer a GUI to guide users through this process.

The private keys are owned by different parties, and I don't see how the subkey concept would help here? Its goal, as I understand it, is to make rotating keys easier. But upgrade keys can already be rotated by upgrading and replacing the burnt-in public key. Attestation keys, on the other hand, include some privacy risk when rotating.

snowyu, 2021-11-25:
I 

> The per-account keys (is this what you mean with FIDO keys?)

Yes.

> The attestation key is a private key that can be configured into OpenSK.

Is the attestation key used to sign the per-account key?  If it is, it is the device key(the second subkey) I am talking about.

> The upgrade key is used in our new, experimental method to upgrade OpenSK through the app. 

The upgrade key should be `the first subkey` . 

`The master key` can be used to revoke/renew the upgrade key and the attestation key.

kaczmarczyck, 2021-11-25:
During MakeCredential, the attestation key can be used to sign the new key material, if using the batch attestation mode.

I'm still not convinced that this structure would make key management simpler for vendors. The attestation key does not benefit from having a primary key, since it is shipped on the device, with the idea that it is too hard to extract from the hardware for an attacker. A primary key doesn't help with that story I think.

The upgrade key on the other hand needs to be kept in the guarded basement (refering to the linked article), even if it is secondary, or an attacker could forge an upgrade an take over control of your device.

snowyu, 2021-12-01:
> During MakeCredential, the attestation key can be used to sign the new key material, if using the batch attestation mode.

The key may be leaked or time-limited. If there is no signature, how to revoke the key?

> I'm still not convinced that this structure would make key management simpler for vendors. 

The signature means more security. If wanna be easier to use, then there should be a supporting GUI program.

> The upgrade key on the other hand needs to be kept in the guarded basement (refering to the linked article), even if it is secondary, or an attacker could forge an upgrade an take over control of your device.

Yes, `the upgrade key` should be never stored on the device, and only the public part of `the upgrade key` is on the device.

It is precisely considering that the upgrade key may be leaked (requires batch operation to upgrade) or become invalid, so there is a master key. According to the OpenPGP specification, the master key itself can also be renewed.

kaczmarczyck, 2021-12-21:
> The key may be leaked or time-limited. If there is no signature, how to revoke the key?

You would need to contact FIDO Alliance to update their MDS (see [their website](https://fidoalliance.org/metadata/) for more info) for revocation.

> The signature means more security. If wanna be easier to use, then there should be a supporting GUI program.

Then maybe I don't understand what you are proposing. Who can see these signatures, who checks them?

> Yes, `the upgrade key` should be never stored on the device, and only the public part of `the upgrade key` is on the device.

This is already the case. I can see a small benefit to having a primary key here. Now, a new upgrade publi key is checked with the old upgrade private key. So if the upgrade private key is leaked, we'd have to be faster to upgrade than an attacker.
If we instead have a primary, the new upgrade key be checked using the primary public key that is stored on device (like the secondary public key). Basically, upgrading the key material and upgrading the firmware become 2 separate processes.

I'd say that makes it a bit more complicated, with the advantage that it is a easier to recover from a loss of the secondary key. We do not plan to ship upgrade very often though, so safely storing the private key in both cases shouldn't be too much work. Also, loss of the upgrade key, even if it can be replaced later, still means that the attacker can export all per-site key material.