google/OpenSK #301: configure.py throws a TypeError when attempting to send keys to USB dongle

Dreamurrrr, 2021-04-08:
## Expected Behavior

For the keys to be sent to the USB device

## Actual Behavior

[image]

Output in text for convenience:
```bash
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK/OpenSK-virtualenv$ ./tools/configure.py \
>     --certificate=crypto_data/opensk_cert.pem \
>     --private-key=crypto_data/opensk.key
info: Private key is valid.
info: Certificate is valid.
Traceback (most recent call last):
  File "./tools/configure.py", line 206, in <module>
    main(parser.parse_args())
  File "./tools/configure.py", line 135, in main
    for authenticator in tqdm(get_opensk_devices(args.batch)):
  File "./tools/configure.py", line 67, in get_opensk_devices
    if (dev.descriptor["vendor_id"],
TypeError: tuple indices must be integers or slices, not str
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK/OpenSK-virtualenv$ 
```

## Steps to Reproduce the Problem

1. Run:
```bash
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
```

2. See error

## Specifications

- Python version: 3.8.5.final.0
- Version: "develop" branch 6216a32
- Distro: Linux Mint 20.1 Ulyssa base: Ubuntu 20.04 focal 

##Extra information that may be helpful

-[My system info](https://gist.github.com/Dreamurrrr/0cb88d306fca0eaab1ea58d7741f2cb9)
-[Packages that have the name "Python" in it.](https://gist.github.com/Dreamurrrr/d8bf538ee29b46ec9749aa0f0d273e54)

kaczmarczyck, 2021-04-08:
Hi! Can you tell me what your version of `libfido2-dev` is? The type of `dev.descriptor` depends on that library's version IIRC.

jmichelp, 2021-04-08:
According to the provided logs, you're using version `0.8.1` of the `fido2` python package (`python3-fido2/focal,focal 0.8.1-1 all`).
You need to upgrade it to version `0.9` or above by running `pip3 install --upgrade 'fido2==0.9.1'` inside your venv.

kaczmarczyck, 2021-04-08:
Oh, missed the provided package list. Thanks!

Dreamurrrr, 2021-04-08:
> According to the provided logs, you're using version `0.8.1` of the `fido2` python package (`python3-fido2/focal,focal 0.8.1-1 all`).
> You need to upgrade it to version `0.9` or above by running `pip3 install --upgrade 'fido2==0.9.1'` inside your venv.

@jmichelp @kaczmarczyck 

I ran `pip3 install --upgrade 'fido2==0.9.1'` in my venv but it already states that I have the packages. 

```bash
notime@notime:~/Desktop/OpenSK-Dev/OpenSK/OpenSK-virtualenv$ source bin/activate
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK-Dev/OpenSK/OpenSK-virtualenv$ pip3 install --upgrade 'fido2==0.9.1'
Requirement already up-to-date: fido2==0.9.1 in ./lib/python3.8/site-packages (0.9.1)
Requirement already satisfied, skipping upgrade: cryptography>=1.5 in ./lib/python3.8/site-packages (from fido2==0.9.1) (3.4.7)
Requirement already satisfied, skipping upgrade: six in ./lib/python3.8/site-packages (from fido2==0.9.1) (1.15.0)
Requirement already satisfied, skipping upgrade: cffi>=1.12 in ./lib/python3.8/site-packages (from cryptography>=1.5->fido2==0.9.1) (1.14.5)
Requirement already satisfied, skipping upgrade: pycparser in ./lib/python3.8/site-packages (from cffi>=1.12->cryptography>=1.5->fido2==0.9.1) (2.20)
```
I then attempted to reinstall python3-fido2, and by accident I typed in `sudo apt install python3-fido2` and for some reason it worked????
```bash
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK-Dev/OpenSK/OpenSK-virtualenv$ sudo apt install python3-fido2
[sudo] password for notime:                  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libu2f-udev
The following NEW packages will be installed:
  libu2f-udev python3-fido2
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 116 kB of archives.
After this operation, 536 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://archive.ubuntu.com/ubuntu focal/main amd64 libu2f-udev all 1.1.10-1 [6,108 B]
Get:2 http://archive.ubuntu.com/ubuntu focal/universe amd64 python3-fido2 all 0.8.1-1 [110 kB]
Fetched 116 kB in 1s (90.6 kB/s)        
Selecting previously unselected package libu2f-udev.
(Reading database ... 363399 files and directories currently installed.)
Preparing to unpack .../libu2f-udev_1.1.10-1_all.deb ...
Unpacking libu2f-udev (1.1.10-1) ...
Selecting previously unselected package python3-fido2.
Preparing to unpack .../python3-fido2_0.8.1-1_all.deb ...
Unpacking python3-fido2 (0.8.1-1) ...
Setting up python3-fido2 (0.8.1-1) ...
Setting up libu2f-udev (1.1.10-1) ...
```
I then attempted to try the command but to no luck...
```bash
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK-Dev/OpenSK/OpenSK-virtualenv$ ./tools/configure.py \
>     --certificate=crypto_data/opensk_cert.pem \
>     --private-key=crypto_data/opensk.key
info: Private key is valid.
info: Certificate is valid.
Traceback (most recent call last):
  File "./tools/configure.py", line 206, in <module>
    main(parser.parse_args())
  File "./tools/configure.py", line 135, in main
    for authenticator in tqdm(get_opensk_devices(args.batch)):
  File "./tools/configure.py", line 67, in get_opensk_devices
    if (dev.descriptor["vendor_id"],
TypeError: tuple indices must be integers or slices, not str
```
I then removed it and tried the configure.py command again (I ran autoremove after removing the package).
```bash
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK-Dev/OpenSK/OpenSK-virtualenv$ sudo apt remove python3-fi
python3-fido2     python3-filelock  
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK-Dev/OpenSK/OpenSK-virtualenv$ sudo apt remove python3-fido2 
[sudo] password for notime:                  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
[... 40 lines cut ...]
Reading state information... Done
Package 'python3-fido2' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
I then decided to try to remove libfido2-1, but it also removed some things along with it, so I just decided it would be better to tell you what I tried so far:
```bash
(OpenSK-virtualenv) notime@notime:~/Desktop/OpenSK-Dev/OpenSK/OpenSK-virtualenv$ sudo apt purge libfido2-1 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libcbor0.6
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  ncurses-term
The following packages will be REMOVED:
  libfido2-1* openssh-client* ssh-askpass-gnome*
The following NEW packages will be installed:
  ncurses-term
0 upgraded, 1 newly installed, 3 to remove and 1 not upgraded.
Need to get 249 kB of archives.
After this operation, 255 kB disk space will be freed.
Do you want to continue? [Y/n] n
Abort.
```

Dreamurrrr, 2021-04-10:
Okay, looks like I blundered a bit 😬 
The package list wasn't actually the packages installed on my system, so that python3-fido2 package was actually never installed, along with the fido libraries.

I just put an updated link to all of my apt packages in the first post, or you can just click [here](https://gist.github.com/Dreamurrrr/d42fe4956ee8495c05bd625c7855b54d) to see it.

So as of currently, the fido pip package is at its latest version and the python3-fido2 package was never actually installed. Just to make sure that I wasn't messing anything up, I re-cloned the developer branch and used pipenv instead. The result was the same.

jmichelp, 2021-04-13:
It seems that I fixed this bug a while ago in the `bugfix` branch with this commit https://github.com/google/OpenSK/commit/958d7a29dc8ab84da7e4f6eae1a97d45e78e04e6

But I guess we forgot to merge this bugfix into both `stable` and `develop`.

PR #304 (2021-04-13): Bugfix
Fixes #301

Fixes the new API for python3 fido2 package and some documentation.
PR #305 (2021-04-13): Bugfix
Fixes #301 

Backports the fix for stable for python3-fido2 API change and some documentation modifications.
closed 2021-04-13 by jmichelp