google/OpenSK #307: fatal: Certificate validity starts in the future.

Samthebest999, 2021-04-14:
## Expected Behavior
When running 
```shell
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
``` 
It is expected to inject the cryptographic data into the Feitian usb dongle.
but it is not and is returning 
```shell
info: Private key is valid.
fatal: Certificate validity starts in the future.
```
## Actual Behavior
```shell
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
info: Private key is valid.
fatal: Certificate validity starts in the future.
```

## Steps to Reproduce the Problem

1. git clone https://github.com/google/OpenSK.git
1. cd OpenSK/
1. ./setup.sh
1. ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
1. start flashing
1. /tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
1. 
## Specifications

- Version: Latest (Cloned Today)
- Platform: Raspberry Pi 4B runnung Raspberry Pi OS Latest

jmichelp, 2021-04-14:
Can you run the following commands so that we can see if there's any discrepancy in the dates:
```sh
$ openssl x509 -in crypto_data/opensk_cert.pem -noout -text
$ date
```

If they are both correct, this can be a timezone issue. Maybe we're comparing UTC with localtime somewhere in our script.

Dreamurrrr, 2021-04-14:
I had this issue too. For some reason when the certificate is generated it makes it valid tomorrow, not on the day it was generated on. This can be observered by running the setup script and then looking at the opensk_cert.pem "not valid before" date.

The workaround I did is to wait for tomorrow or the day after and then push the keys to the dongle.

Samthebest999, 2021-04-15:
I tried that this is what i  got
```shell
Certificate:
    Data:
        Version: 1 (0x0)
        Serial Number:
            3b:49:e3:1e:7e:0d:5b:21:7c:3b:40:1e:38:7c:48:fd:20:74:0e:2f
        Signature Algorithm: ecdsa-with-SHA256
        Issuer: CN = OpenSK CA
        Validity
            Not Before: Apr 15 12:45:40 2021 GMT
            Not After : Apr 15 12:45:40 2031 GMT
        Subject: CN = OpenSK Hacker Edition
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub:
                    04:54:4f:de:63:71:df:72:cc:a3:c2:62:f1:db:fb:
                    df:fb:5b:77:54:0b:ed:24:28:7f:d5:f5:f0:87:fa:
                    d0:67:16:99:04:ae:fb:8e:03:a9:01:87:4d:55:e5:
                    8a:72:66:19:87:cd:d3:46:0a:65:16:d7:2e:45:d0:
                    4a:2a:a6:36:63
                ASN1 OID: prime256v1
                NIST CURVE: P-256
    Signature Algorithm: ecdsa-with-SHA256
         30:45:02:21:00:bc:d8:ad:8c:6b:17:1e:2f:ba:e2:99:af:53:
         0b:2b:05:f8:a5:c8:4c:4d:bf:69:e5:0e:a5:c0:23:28:17:2c:
         42:02:20:2a:6c:14:e4:da:d7:e3:55:f0:c2:d1:d4:c2:fa:ee:
         cd:72:ec:53:9b:2e:6b:f9:1b:c7:37:b6:58:83:06:fe:8f
```
Then I ran
 ```shell
date
```
Output:
```shell
Thu 15 Apr 2021 08:00:12 AM CDT
```

jmichelp, 2021-04-15:
Thanks.
If I'm correct 8am CDT is 1pm GMT. And the certificate is valid from 12:45am GMT. So this looks fine.
Which means that the python script is probably comparing the timestamps without caring about the timezones.
I'll make some tests.

jmichelp, 2021-04-15:
Could you try changing the line 116 of `configure.py` script from
```py
    now = datetime.datetime.now()
```
to
```py
    now = datetime.datetime.utcnow()
```

OpenSSL seems to always return timestamp as UTC and we were indeed comparing with local time. This should ensure we always compare UTC timestamps.

Samthebest999, 2021-04-15:
OK, Will do

Samthebest999, 2021-04-15:
Yep! Now it Works!!

PR #308 (2021-04-15): Compare all timestamps using UTC timezone
Fixes #307 

- [x] Tests pass
- [ ] Appropriate changes to README are included in PR
PR #309 (2021-04-15): Compare all timestamps in UTC timezone.
Fixes #307 

- [x] Tests pass
- [ ] Appropriate changes to README are included in PR
closed 2021-04-15 by jmichelp