google/OpenSK #313: Communicating with OpenSK over USB HID control pipe.

alishir, 2021-05-15:
## Expected Behavior
Being able to communicate with OpenSK dongle over HID control pipe.

## Actual Behavior
OpenSK does not reply CTAPHID_INIT over control pipe.

## Steps to Reproduce the Problem

1. Program dongle
2. Send CTAPHID_INIT packet over control pipe
3. Get invalid response

## Specifications

- Version: dev branch
- Platform: Debian testing

jmichelp, 2021-05-17:
Thanks Ali for reporting.

Do I understand correctly that you're trying to send a CTAP_INIT packet to the USB control endpoint (i.e. EP0)?
If that's the case, then the code is working as intended. HID communication happens on interrupt endpoints (EP1 in OpenSK case). Control endpoints are only used to enumerate/configure the USB device (e.g. getting descriptors, switching between configurations, etc.).
I reckon that other FIDO2 authenticators would behave exactly the same.

Could you clarify exactly what you were doing so that we could reproduce and confirm whether the behavior is intended or not?

alishir, 2021-05-18:
Thanks for your comment. According to the HID specification, HID devices could communicate with the host via a control pipe on an interrupt pipe. In UEFI environment host only could receive data over the interrupt pipe and could not send data to the HID device through the interrupt pipe[1]. A simple solution for this limitation is sending HID packets over the control pipe, I've tested this approach with qemu [u2f emulated](https://github.com/qemu/qemu/blob/master/docs/u2f.txt) device with a simple patch that forwards some control pipe packets over the interrupt pipe.

Also, I've implemented a simple UEFI application that sends CTAP_INIT over the control pipe and receives the response over interrupt pipe, but when I pass-through the OpenSK device to the qemu and launch that UEFI it does not work. I've captured the USB packets and it seems that OpenSK does not process CTAP_INIT over the control pipe so it does not send any response.

I've read the OpenSK usb patch and it seems it handles control packets only for device enumeration and configuration through `ClientCtrl`, if we could pass other packets to the OpenSK core we could use OpenSK in more restricted environments such as UEFI.

[1] https://github.com/tianocore/edk2/commit/e172ea16a7002f736a0c3d39c127a890f120d374

jmichelp, 2021-05-18:
The Nordic chip we're using, just like many other microcontrollers, only support 1 control endpoint IN+OUT and by USB specification EP0 must be Control (otherwise your device simply won't work).

On top of that, FIDO2 specification clearly indicates that the 2 required endpoints used for USB HID transport are of type Interrupt.
What you're currently trying to do goes pretty much against (or let's say in a gray area of) 2 specifications out of 3. So it might work but probably not in a stable way across platforms. And we can't start supporting hacks like this in our codebase unfortunately.

alishir, 2021-05-18:
Thanks a lot for your explanation. You are right, so let me close this issue.