google/OpenSK #314: Keeping credentials after re-flashing?

axmmisaka, 2021-05-18:
This might be a question related to how FIDO works instead, but I guess it might be related to OpenSK as there's https://github.com/google/OpenSK/issues/179.  
I am using a `nrf52840_mdk_dfu`, and it used to be programmed with an older version of OpenSK. I fetched and merged the current stable branch, leaving contents in `crypto_data/` untouched. After I clear the persistent storage (I cleared the storage with Nordic nRF Connect) and flashed my dongle with JLink/UF2 bootloader, it appears that Windows Credential Manager/websites does not recognise this key anymore.   
My questions are
1. Is it an expected behaviour that after clearing persistent storage and re-programming the dongle with OpenSK built with same content in `crypto_data/`, applications don't recognise it anymore?
2. If not, what could possibly go wrong and cause this problem? Is it because of the version update or because I possibly incorrectly operated on `crypto_data/`?

Thank you for your time.

kaczmarczyck, 2021-05-18:
From the title, I guess that
> Windows Credential Manager/websites does not recognise this key anymore

means you lost your credentials, not that OpenSK is generally unusable and unable to create new credentials?

If my interpretation is correct, then everything is working as intended. Since you cleared the persistent storage, you lost your credentials. None of your credentials' private keys are stored in `crypto_data/`. Else you wouldn't need an extra piece of hardware. If you started with an old version of OpenSK (before we switched to our storage v2), we don't support upgrading without credential loss. Starting with the certified stable firmware, you should be able to flash without `--clear_storage`.

I think your question comes from a general misconception of what `crypto_data/` is for? We use it to initialise OpenSK with device model identifiers and attestation keys. Let me know if this was helpful or if you have more questions!

axmmisaka, 2021-05-19:
Got it! Thank you for the speedy reply.
To confirm, is it fair to say that, starting with the certified stable firmware, as long as `--clear_storage` flag is not added, the storage will not be cleared (even if we flash with IntelHex in a proper manner)?

kaczmarczyck, 2021-05-19:
> To confirm, is it fair to say that, starting with the certified stable firmware, as long as `--clear_storage` flag is not added, the storage will not be cleared (even if we flash with IntelHex in a proper manner)?

I could image future changes that require clearing the storage, like changing the memory layout. All changes and flashing methods that don't touch the memory region with the persistent storage in flash should be safe. If we ever have to make one of these changes, we'd document them and warn our users.

> Further (I might be asking too much), I wonder if you could kindly guide me which code are relevant to generating credentials (which I assume was done when OpenSK is run for the first time)? Thank you!

There is a lot of code that is relevant for generating credentials. The command [`MakeCredential`](https://fidoalliance.org/specs/fido-v2.1-rd-20210309/fido-client-to-authenticator-protocol-v2.1-rd-20210309.html#authenticatorMakeCredential) can eventually lead to us generating a resident or a server-side credential. When OpenSK is run for the first time, we initialise and store master secrets that encrypt the server-side credentials. You'd have to make your question more specific so I can answer it properly?

axmmisaka, 2021-05-19:
Sorry for the ambiguous question, I was wondering which code are responsible for initialising and storing the master secret when OpenSK run for the first time.

kaczmarczyck, 2021-05-19:
That would be [here](https://github.com/google/OpenSK/blob/5e682d9e176e936c22fcb963a708ffb0b47a33e6/src/ctap/storage.rs#L105).

axmmisaka, 2021-05-19:
I very much appreciate your rapid help and my apologies for any disturbance caused.

closed 2021-05-19 by axmmisaka