google/OpenSK #316: --no-u2f flag broken

kaczmarczyck, 2021-06-09:
The flag to deploy without U2F is broken.

## Expected Behavior
OpenSK is deployed.

## Actual Behavior
```sh
Traceback (most recent call last):
  File "/usr/local/google/home/kaczmarczyck/fido2/OpenSK/./deploy.py", line 918, in <module>
    partial_args, _ = main_parser.parse_known_args()
  File "/usr/lib/python3.9/argparse.py", line 1863, in parse_known_args
    namespace, args = self._parse_known_args(args, namespace)
  File "/usr/lib/python3.9/argparse.py", line 2072, in _parse_known_args
    start_index = consume_optional(start_index)
  File "/usr/lib/python3.9/argparse.py", line 2012, in consume_optional
    take_action(action, args, option_string)
  File "/usr/lib/python3.9/argparse.py", line 1940, in take_action
    action(self, namespace, argument_values, option_string)
  File "/usr/local/google/home/kaczmarczyck/fido2/OpenSK/./deploy.py", line 250, in __call__
    if self.const in items:
TypeError: argument of type 'NoneType' is not iterable
```

## Steps to Reproduce the Problem

Setup as usual, then deploy with:
```sh
./deploy.py --board=nrf52840dk --no-u2f
```

## Specifications

- Version: develop
- Platform: Linux

jmichelp, 2021-06-23:
@kaczmarczyck can you confirm this has been fixed?

kaczmarczyck, 2021-06-23:
Confirmed, works.

PR #317 (2021-06-09): Fix broken parsing.
By setting the default value before pre-parsing we ensure that the item
can't be None. As an extra safety the custom action also checks for
None.

Fixes #316 

- [x] Tests pass
- [ ] Appropriate changes to README are included in PR
PR #323 (2021-06-09): Fix broken parsing
By setting the default value before pre-parsing we ensure that the item
can't be None. As an extra safety the custom action also checks for
None.

Fixes #316 on develop branch

- [x] Tests pass
- [ ] Appropriate changes to README are included in PR
closed 2021-06-23 by kaczmarczyck