google/OpenSK #320: Testing the persistent store library in github workflows now fails

jmichelp, 2021-06-09:
## Expected Behavior

Tests should pass if the code is left unchanged by a PR

## Actual Behavior

Tests fail because the toolchain is not properly set up.

## Steps to Reproduce the Problem

1. Create a PR which doesn't affect the persistent store.

## Specifications

- Version: N/A
- Platform: N/A

kaczmarczyck, 2021-06-09:
The fix in #321 mentions it is temporary and needs more work.
@ia0 keep this open?

kaczmarczyck, 2021-06-09:
Ah, I just saw #322, I assume this is what we actually want.

jmichelp, 2021-06-23:
@ia0 This has been fixed, right? Or is there anything pending?

ia0, 2021-06-23:
Ah yes sorry, I thought that writing "Fixes #320" in the description of #322 would automatically close this. But apparently I was wrong. Closing manually then.

closed 2021-06-23 by ia0