google/OpenSK #337: Add code coverage tracking

ebursztein, 2021-07-01:
## Expected Behavior

Have each merge request produce a code coverage report tracked over time via https://coveralls.io/github/google/

## Actual Behavior

## Steps to Reproduce the Problem

1.
1.
1.

## Specifications

- Version:
- Platform:

jmichelp, 2021-07-01:
I did a quick investigation.
Corresponding Github actions exist so this doesn't seem too hard.
I configured and enabled coveralls for this repository.

What's next:
- [x] write Github workflow
- [ ] write grcov configuration
- [ ] test
- [ ] create PR

Rust code should be compiled with specific options and unfortunately, this doesn't seem to compile:
```
  = note: clang: warning: argument unused during compilation: '-no-pie' [-Wunused-command-line-argument]
          Undefined symbols for architecture x86_64:
            "_libtock_alloc_init", referenced from:
                _rust_start in liblibtock_core-7e139524ee594146.rlib(libtock_core-7e139524ee594146.libtock_core.7k5hr3h7-cgu.0.rcgu.o)
          ld: symbol(s) not found for architecture x86_64
          clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

jmichelp, 2021-07-05:
The `develop` branch is covered by the fix but the badge doesn't point to the correct branch.
And it needs a follow-up PR for the `stable` branch

PR #339 (2021-07-05): Coveralls
Fixes #337 

A separate PR will be done to add code coverage to the `stable` branch too.

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
PR #342 (2021-07-06): Coveralls workflow applied also to stable
Fixes #337 

This PR consist of:
- cherry picking the 2 commits pushed to `develop` branch
- moving all the branches for the badges to `stable` 

- [x] Tests pass
- [x] Appropriate changes to README are included in PR
closed 2021-07-06 by jmichelp