google/OpenSK #343: launchpadlib 1.10.13 is not installed during setup

mxi1, 2021-07-06:
## Expected Behavior
The `lunchpadlib` should be installed.  Can I install the latest lanuchpadlib instead of the specific version "1.10.13"?  Thanks.

## Actual Behavior

> ERROR: launchpadlib 1.10.13 requires testresources, which is not installed.

## Steps to Reproduce the Problem

1.  run the default `setup.sh`, you will see this error.

## Specifications

- Version: the stable version, and the latest commit is 7e5e2a665ff4f9ba68b7e8d098ee2248a3a85bb8
- Platform: Ubuntu 20.04 x86_64

kaczmarczyck, 2021-07-06:
Hi!

I don't find any mention of `launchpadlib` (or `lunchpadlib`, but that's probably a typo?) in our code, or a requirement of version  "1.10.13". Can you point me to where this comes from?

mxi1, 2021-07-07:
Okay, thanks. as you replied, it seems to be one dependency for one pip package required by your shell script. 
I cannot find the previous shell output right now, need to re-run your `setup.sh` script on another clean machine, and I will post the update here. Thanks. :smile:

jmichelp, 2021-08-16:
@mxi1 Any update?

mxi1, 2021-08-17:
> @mxi1 Any update?
It's still there. I googled the error message, and found that the version of `launchpadlib`  is the root cause. Instead, the `testresources` package is.

In order to fix this issue, the `python3-testresources` or `python-testresources` is required before running `./setup.sh`.

mxi1, 2021-08-17:
[ERROR: launchpadlib 1.10.6 requires testresources, which is not installed ](https://gist.github.com/y56/0540d22a1db40dacc7fbbb93c866821e) is one reference hosted on github gist website.

kaczmarczyck, 2021-09-07:
I was unable to reproduce on a freshly installed Debian-based distro.
This seems like a packaging bug in Ubuntu / Debian to me, that might have been fixed? If you think a workaround for your Ubuntu version is worthwhile, feel free to contribute one.

mxi1, 2021-09-08:
Okay, let me have a try.

mxi1, 2021-09-10:
> I was unable to reproduce on a freshly installed Debian-based distro.
> This seems like a packaging bug in Ubuntu / Debian to me, that might have been fixed? If you think a workaround for your Ubuntu version is worthwhile, feel free to contribute one.

Please check this PR: https://github.com/google/OpenSK/pull/376

kaczmarczyck, 2021-09-10:
Feel free to close this issue if it's resolved on bugfix!

PR #376 (2021-09-10): Added pypi:testresources required by one of the tockloader dependencies;

Fixes #343

> It's a good idea to open an issue first for discussion.

Running `setup.sh` on ubuntu versions (at least 20.04, 20.10),  there will be one unmet dependency error message, indicating "launchpadlib" requires "testresources". 

The pypi package "testresources" requires to be installed, but it doesn't. 
 
The solution is easy here: install testresouces pypi package, before installing "tockloader", "intelhex", etc.

- [Y] Tests pass
- [ ] Appropriate changes to README are included in PR
closed 2021-09-11 by mxi1