google/OpenSK #346: Content are not match or out to date

ball-spongebob, 2021-07-17:
Please review, edit or update [install.md](https://github.com/google/OpenSK/blob/stable/docs/install.md) and [readme](https://github.com/google/OpenSK/blob/stable/README.md) content is not match or not update in path sector "OpenSK" (compatible devices)

kaczmarczyck, 2021-07-19:
Hi, thanks for reaching out. Can you provide more details on where you think the documentation is lacking or out of date? What branch are you on?

ball-spongebob, 2021-08-02:
[image]
[image]
P.S.Why do you care "Branch"?! It so Stupid! Fool! idiot!🙄

closed 2021-08-02 by kaczmarczyck