google/OpenSK #371: Setup fails with error compiling clap in macOS

daviddesmet, 2021-08-15:
## Expected Behavior
`setup.sh` run successfully.

## Actual Behavior
`setup.sh` doesn't complete due to error while compiling clap.

## Steps to Reproduce the Problem

1. Execute `./setup.sh`
Produces the following output:

```bash
[-] Copying additional boards to Tock... DONE.
[-] Applying patch "01-persistent-storage.patch"... DONE.
[-] Applying patch "02-usb.patch"... DONE.
[-] Applying patch "03-usb-debugging.patch"... DONE.
[-] Applying patch "04-additional-boards.patch"... DONE.
[-] Applying patch "05-mpu-fix.patch"... DONE.
[-] Applying patch "06-update-uicr.patch"... DONE.
[-] Applying patch "07-firmware-protect.patch"... DONE.
Signature ok
subject=/CN=OpenSK CA
Getting Private key
Signature ok
subject=/CN=OpenSK Hacker Edition
Getting CA Private Key
info: syncing channel updates for 'nightly-2020-06-10-x86_64-apple-darwin'

  nightly-2020-06-10-x86_64-apple-darwin unchanged - rustc 1.46.0-nightly (feb3536eb 2020-06-09)

info: checking for self-updates
Requirement already satisfied: tockloader==1.5 in /Library/Python/3.8/site-packages (1.5.0)
Requirement already satisfied: six in /Users/dds/Library/Python/3.8/lib/python/site-packages (1.16.0)
Requirement already satisfied: intelhex in /Library/Python/3.8/site-packages (2.3.0)
Requirement already satisfied: pytoml>=0.1.11 in /Library/Python/3.8/site-packages (from tockloader==1.5) (0.1.21)
Requirement already satisfied: colorama>=0.3.7 in /Library/Python/3.8/site-packages (from tockloader==1.5) (0.4.4)
Requirement already satisfied: pyserial>=3.0.1 in /Library/Python/3.8/site-packages (from tockloader==1.5) (3.5)
Requirement already satisfied: argcomplete>=1.8.2 in /Library/Python/3.8/site-packages (from tockloader==1.5) (1.12.3)
Requirement already satisfied: crcmod>=1.7 in /Library/Python/3.8/site-packages (from tockloader==1.5) (1.7)
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
    Updating crates.io index
  Installing elf2tab v0.6.0
   Compiling libc v0.2.99
   Compiling proc-macro2 v1.0.28
   Compiling version_check v0.9.3
   Compiling autocfg v1.0.1
   Compiling unicode-xid v0.2.2
   Compiling syn v1.0.74
   Compiling unicode-segmentation v1.8.0
   Compiling unicode-width v0.1.8
   Compiling ansi_term v0.11.0
   Compiling cfg-if v1.0.0
   Compiling bitflags v1.3.1
   Compiling vec_map v0.8.2
   Compiling strsim v0.8.0
   Compiling byteorder v0.5.3
   Compiling lazy_static v1.4.0
   Compiling heck v0.3.3
   Compiling textwrap v0.11.0
   Compiling num-traits v0.2.14
   Compiling num-integer v0.1.44
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling elf v0.0.10
   Compiling quote v1.0.9
   Compiling atty v0.2.14
   Compiling time v0.1.44
   Compiling filetime v0.2.15
   Compiling xattr v0.2.2
   Compiling clap v2.33.3
error[E0658]: `if` is not allowed in a `const fn`
  --> /Users/david/.cargo/registry/src/github.com-1ecc6299db9ec823/clap-2.33.3/src/app/settings.rs:7:1
   |
7  | / bitflags! {
8  | |     struct Flags: u64 {
9  | |         const SC_NEGATE_REQS       = 1;
10 | |         const SC_REQUIRED          = 1 << 1;
...  |
51 | |     }
52 | | }
   | |_^
   |
   = note: see issue #49146 <https://github.com/rust-lang/rust/issues/49146> for more information
   = help: add `#![feature(const_if_match)]` to the crate attributes to enable
   = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0658]: `if` is not allowed in a `const fn`
  --> /Users/david/.cargo/registry/src/github.com-1ecc6299db9ec823/clap-2.33.3/src/args/settings.rs:6:1
   |
6  | / bitflags! {
7  | |     struct Flags: u32 {
8  | |         const REQUIRED         = 1;
9  | |         const MULTIPLE         = 1 << 1;
...  |
28 | |     }
29 | | }
   | |_^
   |
   = note: see issue #49146 <https://github.com/rust-lang/rust/issues/49146> for more information
   = help: add `#![feature(const_if_match)]` to the crate attributes to enable
   = note: this error originates in a macro (in Nightly builds, run with -Z macro-backtrace for more info)

   Compiling tar v0.4.37
   Compiling chrono v0.4.19
error: aborting due to 2 previous errors

For more information about this error, try `rustc --explain E0658`.
error: could not compile `clap`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: failed to compile `elf2tab v0.6.0`, intermediate artifacts can be found at `/var/folders/sc/c77xmhjn1rx5gffjzxl2p_dm0000gn/T/cargo-installpItNMj`

Caused by:
  build failed
```

## Specifications

- Version: current stable (commit e02eaa2cacb6f2949fe282d9e32129db30dccfbb)
- Platform: nrf52840_dongle
- OS: macOS Catalina 10.15.7

daviddesmet, 2021-08-15:
Just noticed this commit e02eaa2cacb6f2949fe282d9e32129db30dccfbb in stable

```
Remove elf2tab dev-dependency (#366)
We don't use it anymore. Not sure when we used to use it.

Fixes #364

```

which removes the dependency on **elf2tab** which is the one causing me issues, so I gave it a try and removed the dependency on `setup.sh` too:

```sh
# Install dependency to create applications.
mkdir -p elf2tab
cargo install elf2tab --version 0.6.0 --root elf2tab/
```

With this change, the `setup.sh` script now works but `deploy.py` doesn't since it requires **elf2tab** when executing the function `create_tab_file`. This is the command I executed:

```bash
./deploy.py --board=nrf52840_dongle_dfu --programmer=nordicdfu --opensk
```

and the output:

```bash
info: Generating Tock TAB file for application/example ctap2
Traceback (most recent call last):
  File "./deploy.py", line 993, in <module>
    main(main_parser.parse_args())
  File "./deploy.py", line 752, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 636, in run
    self.build_opensk()
  File "./deploy.py", line 357, in build_opensk
    self._build_app_or_example(is_example=False)
  File "./deploy.py", line 393, in _build_app_or_example
    self.create_tab_file({props.arch: app_path})
  File "./deploy.py", line 409, in create_tab_file
    elf2tab_ver = self.checked_command_output(
  File "./deploy.py", line 297, in checked_command_output
    cmd_output = subprocess.run(
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'elf2tab/bin/elf2tab'
```

ia0, 2021-08-15:
Hi @daviddesmet ,

Thanks for the report. Indeed, this issue started happening recently and was fixed in the develop branch with #369. I just created a backport in #372 which you may use to fix your issue until it's merged. The #366 commit you mention is actually unrelated although mentioning elf2tab too. It was to fix a cargo-audit warning.

daviddesmet, 2021-08-15:
Thanks for the follow-up and for the backport!

I noticed later it was not related to elf2tab when I dug a bit more about it. I didn't expect it was just a matter of switching Rust to the stable compiler for the rust tools (not a Rust developer here)... anyway, I manually applied your changes from `setup.sh` so I don't have to wait for the PR to be merged since I wanted to have the device flashed so I can toy with it.

Thanks again!

PR #372 (2021-09-09): Install Rust tools with stable compiler
We only need the frozen nightly for Tock and the app.

Backport of #369 generated with `git cherry-pick a80ff42`.

FIxes #371.
closed 2021-09-09 by kaczmarczyck