google/OpenSK #389: Google 2-factor authentication does not accept opensk (on nrf_dongle)

dataselfservice, 2021-10-02:
## Expected Behavior

Google accepts nrf_dongle opensk as 2F authentication.

## Actual Behavior

It does not

## Steps to Reproduce the Problem

1. build+deploy latest: https://github.com/google/OpenSK/commit/420f062a2173df9c63dd6133ed72b1dee835d968
2. deployed to nrf_dongle, tested with WebAuthN or Youico -> ok
3. Google|MyAccount|Security|2F devices|Add|USB, opensk led start blinking but after pushing the button, nothin happens

## Specifications

- Version: https://github.com/google/OpenSK/commit/420f062a2173df9c63dd6133ed72b1dee835d968
- Platform: nrf_dongle

ia0, 2021-10-02:
Thanks for the report! I could reproduce. Here's the output of `./deploy.py --board=nrf52840dk --opensk --debug --panic-console --clear-storage` on the stable branch.

```
Bus reset
Bus reset
Received packet at 3.093505 s
Received message: Message { cid: [ff, ff, ff, ff], cmd: 06, payload: [1b, 2c, 85, af, 7b, f7, aa, 39] }
Sending message: Message { cid: [ff, ff, ff, ff], cmd: 06, payload: [1b, 2c, 85, af, 7b, f7, aa, 39, 00, 00, 00, 01, 02, 01, 00, 00, 05] }
Sent packet at 3.105346 s
Received packet at 3.109497 s
Received message: Message { cid: [00, 00, 00, 01], cmd: 10, payload: [04] }
Received command: Ok(
    AuthenticatorGetInfo,
)
Sending response: Ok(
    AuthenticatorGetInfo(
        AuthenticatorGetInfoResponse {
            versions: [
                "U2F_V2",
                "FIDO_2_0",
            ],
            extensions: Some(
                [
                    "hmac-secret",
                ],
            ),
            aaguid: [
                54,
                45,
                91,
                116,
                40,
                112,
                71,
                249,
                172,
                138,
                155,
                139,
                2,
                76,
                153,
                176,
            ],
            options: Some(
                {
                    "clientPin": false,
                    "rk": true,
                    "up": true,
                },
            ),
            max_msg_size: Some(
                1024,
            ),
            pin_protocols: Some(
                [
                    1,
                ],
            ),
            default_cred_protect: None,
        },
    ),
)
Sending message: Message { cid: [00, 00, 00, 01], cmd: 10, payload: [00, a6, 01, 82, 66, 55, 32, 46, 5f, 56, 32, 68, 46, 49, 44, 4f, 5f, 32, 5f, 30, 02, 81, 6b, 68, 6d, 61, 63, 2d, 73, 65, 63, 72, 65, 74, 03, 50, 36, 2d, 5b, 74, 28, 70, 47, f9, ac, 8a, 9b, 8b, 02, 4c, 99, b0, 04, a3, 62, 72, 6b, f5, 62, 75, 70, f5, 69, 63, 6c, 69, 65, 6e, 74, 50, 69, 6e, f4, 05, 19, 04, 00, 06, 81, 01] }
Sent packet at 3.153350 s
Sent packet at 3.157348 s
Received packet at 3.161499 s
Received packet at 3.165496 s
Received message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [00, 01, 03, 00, 00, 00, 40, 52, 49, 32, 9c, f0, ab, d0, ea, ba, 33, 5a, 86, 2c, c7, 70, b8, da, 58, 3b, 75, 59, 95, 74, 3a, f3, a8, fe, 7b, f6, c6, 6d, 8f, a5, 46, 72, b2, 22, c4, cf, 95, e1, 51, ed, 8d, 4d, 3c, 76, 7a, 6c, c3, 49, 43, 59, 43, 79, 4e, 88, 4f, 3d, 02, 3a, 82, 29, fd, 00, 00] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [69, 85] }
[... 9 lines cut ...]
Sending message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [69, 85] }
Sent packet at 3.629364 s
Received packet at 3.833465 s
Received packet at 3.837493 s
Received message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [00, 01, 03, 00, 00, 00, 40, 52, 49, 32, 9c, f0, ab, d0, ea, ba, 33, 5a, 86, 2c, c7, 70, b8, da, 58, 3b, 75, 59, 95, 74, 3a, f3, a8, fe, 7b, f6, c6, 6d, 8f, a5, 46, 72, b2, 22, c4, cf, 95, e1, 51, ed, 8d, 4d, 3c, 76, 7a, 6c, c3, 49, 43, 59, 43, 79, 4e, 88, 4f, 3d, 02, 3a, 82, 29, fd, 00, 00] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [69, 85] }
Sent packet at 3.853332 s
Received packet at 4.057464 s
Received packet at 4.061462 s
Received message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [00, 01, 03, 00, 00, 00, 40, 52, 49, 32, 9c, f0, ab, d0, ea, ba, 33, 5a, 86, 2c, c7, 70, b8, da, 58, 3b, 75, 59, 95, 74, 3a, f3, a8, fe, 7b, f6, c6, 6d, 8f, a5, 46, 72, b2, 22, c4, cf, 95, e1, 51, ed, 8d, 4d, 3c, 76, 7a, 6c, c3, 49, 43, 59, 43, 79, 4e, 88, 4f, 3d, 02, 3a, 82, 29, fd, 00, 00] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [69, 85] }
Sent packet at 4.077331 s
Received packet at 4.281463 s
Received packet at 4.285461 s
Received message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [00, 01, 03, 00, 00, 00, 40, 52, 49, 32, 9c, f0, ab, d0, ea, ba, 33, 5a, 86, 2c, c7, 70, b8, da, 58, 3b, 75, 59, 95, 74, 3a, f3, a8, fe, 7b, f6, c6, 6d, 8f, a5, 46, 72, b2, 22, c4, cf, 95, e1, 51, ed, 8d, 4d, 3c, 76, 7a, 6c, c3, 49, 43, 59, 43, 79, 4e, 88, 4f, 3d, 02, 3a, 82, 29, fd, 00, 00] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [69, 85] }
Sent packet at 4.301330 s
Received packet at 4.505462 s
Received packet at 4.509460 s
Received message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [00, 01, 03, 00, 00, 00, 40, 52, 49, 32, 9c, f0, ab, d0, ea, ba, 33, 5a, 86, 2c, c7, 70, b8, da, 58, 3b, 75, 59, 95, 74, 3a, f3, a8, fe, 7b, f6, c6, 6d, 8f, a5, 46, 72, b2, 22, c4, cf, 95, e1, 51, ed, 8d, 4d, 3c, 76, 7a, 6c, c3, 49, 43, 59, 43, 79, 4e, 88, 4f, 3d, 02, 3a, 82, 29, fd, 00, 00] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: 03, payload: [6f, 00] }
Sent packet at 4.621337 s
```

After which everything looks freezed (the board doesn't print anything more but the led next to the chip flashes irregularly) and Google continues to show "Insert your security key and touch it" although it has been touched. When clicking cancel, nothing is additionally printed on the board.

jmichelp, 2021-10-04:
The trace shows that U2F protocol is used. The first few packets return `69 85` which indicates that OpenSK is waiting for a touch. Then the touch happens and OpenSK returns `6f 00` which is an error code.
Unlike CTAP2 protocol, U2F requires that a certificate and its associated private key are injected into the key. This can be achieved using the provided script `tools/configure.py`.

Were you following this extra step?
If yes, we have a regression.

ia0, 2021-10-04:
> Were you following this extra step?

I didn't. I'll need to try again when I have my board.

dataselfservice, 2021-10-04:
> Were you following this extra step?
@jmichelp Actually not. By following those, it works.

kaczmarczyck, 2021-10-05:
Thanks for checking!

Lostsite, 2021-11-12:
I tryed on Nordic nRF52840-dongle, and after I touched the device, it says "can't use this security key, please try other security keys". and run the `./tools/configure.py` shows
'info: Certificate: Missing
info: Private Key: Missing'

@ia0 , @dataselfservice , have you guys faced the same problem? How did you guys solve that? (Sorry I'm still a new hand on the Linux and all these things"

kaczmarczyck, 2021-11-12:
Hi! Did you run the configure tool with the parameters to set the certificate?

```shell
./tools/configure.py --certificate=crypto_data/opensk_cert.pem --private-key=crypto_data/opensk.key
```

If so, this info should change to indicate that you now have a certificate installed.

dataselfservice, 2021-11-12:
> I tryed on Nordic nRF52840-dongle, and after I touched the device, it says "can't use this security key, please try other security keys". and run the `./tools/configure.py` shows 'info: Certificate: [...]

this <https://github.com/google/OpenSK/issues/389#issuecomment-966896651> worked fine (with own certificate/key)

Lostsite, 2021-11-12:
> Hi! Did you run the configure tool with the parameters to set the certificate?
> 
> ```shell
> ./tools/configure.py --certificate=crypto_data/opensk_cert.pem --private-key=crypto_data/opensk.key
> ```
> 
> If so, this info should change to indicate that you now have a certificate installed.

Thanks, I am going back and checking this evening. Tank you!

Lostsite, 2021-11-13:
It works, thank you!

closed 2021-11-13 by kaczmarczyck