google/OpenSK #396: Trouble installing package using a virtual environment

mlubej, 2021-10-30:
Hi! 

I am experiencing the issue described above

## Expected Behavior

After I recursively download the repo I would expect that running `./setup` would run successfully

## Actual Behavior

I get a crash relatively soon on, stating that `Can not perform a '-- user install, user site packages are not visible in this virtualenv?`

I once was able to achieve installing this, but I have no idea what I did differently, so I can't say much about it.

## Steps to Reproduce the Problem

1. I am using `pyenv` and a virtual environment of version `3.8.7`
2. After activating the venv run the `setup.sh` script
3. Observe the crash

## Specifications

- Version: `pyenv 2.2.0`, `python 3.8.7`, `OpenSK stable`
- Platform: macOS Monterey

mlubej, 2021-10-31:
Nevermind, I simply replaced `pip3 install --user` with `pip install`.

Closing.

closed 2021-10-31 by mlubej

kaczmarczyck, 2021-11-01:
Thanks for still writing the detailed bug report, maybe someone stumbles upon this issue and sees this workaround!