google/OpenSK #399: Many tests fail from "u2f reference code"

liamjm, 2021-11-02:
## Expected Behavior
All tests pass for u2f-ref-code/u2f-tests/HID/HIDTest (https://github.com/google/u2f-ref-code)

## Actual Behavior
Several tests fail

## Steps to Reproduce the Problem

1. Clone about repo and `make all` in u2f-ref-code/u2f-tests/HID/
1. Identify HID device to test (eg `u2f-ref-code/u2f-tests/HID/list`)
1. Run tests, eg `HIDTest  /dev/hidraw3 -V` ("-V" log the frames to aid debugging).

The first error is below.
```
t+0.004> deadbeef:86[8]:52A446F93FC94A3800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< deadbeef:bf[1]:0B0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CHECK_EQ fail at test_InitOnNonBroadcastEchoesCID[475]:r.init.cmd != U2FHID_INIT: Aborted
```

The full trace is below.

## Specifications

- Version: debian
- Platform: linux

## Appendix
The "-a" continues after errors.
`./HIDTest  /dev/hidraw3 -V -a`

Output. I haven't looked at each error in detail, but as always, some of the failures might affect later tests.
```
t+0.358< (timeout)
PASS(test_Idle())
t+0.001> ffffffff:86[8]:6840FCA3EBB8F23D00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< ffffffff:86[17]:6840FCA3EBB8F23D000000BB020100000500000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.004> ffffffff:86[8]:D3E68D3FB7660D7C00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< ffffffff:86[17]:D3E68D3FB7660D7C000000BC020100000500000000000000000000000000000000000000000000000000000000000000000000000000000000
PASS(test_Init())
t+0.004> ffffffff:86[8]:22605069E3EB926A0001B818B7BE550000C01606258C7F0000C02406258C7F0000A263F1248C7F000040B728258C7F00000000000000000000
t+0.024< ffffffff:86[17]:22605069E3EB926A000000BD020100000500000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.004> 000000bd:86[8]:BA16B2F1D9C89E4200000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< 000000bd:86[17]:BA16B2F1D9C89E42000000BD020100000500000000000000000000000000000000000000000000000000000000000000000000000000000000
PASS(test_BasicInit())
t+0.004> 000000bd:bc[0]:000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< 000000bd:bf[1]:010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
PASS(test_Unknown(U2FHID_SYNC))
t+0.004> deadbeef:86[8]:099AE5F453D7312600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< deadbeef:bf[1]:0B0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CHECK_EQ fail at test_InitOnNonBroadcastEchoesCID[475]:r.init.cmd != U2FHID_INIT: (continuing -a)
CHECK_EQ fail at test_InitOnNonBroadcastEchoesCID[476]:MSG_LEN(r) != sizeof(U2FHID_INIT_RESP): (continuing -a)
CHECK_EQ fail at test_InitOnNonBroadcastEchoesCID[477]:0 != memcmp(f.init.data, r.init.data, cs): (continuing -a)
CHECK_EQ fail at test_InitOnNonBroadcastEchoesCID[485]:cid != 0xdeadbeef: (continuing -a)
PASS(test_InitOnNonBroadcastEchoesCID())
t+0.004> 000000bd:86[8]:BDBE65752472F13500000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< 000000bd:86[17]:BDBE65752472F135000000BD020100000500000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.004> 000000bd:84[1]:000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< 000000bd:bf[1]:010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
PASS(test_InitUnderLock())
t+0.004> 000000bd:81[99]:7478808ED210E18C26947D005C1B4265B6275909FE8B2FBC4994316E0722A37B9A23096C33EBF95A7F765ADB929C4148C39A51C125807D6F14
t+0.004> 000000bd:86[8]:AEDD1BD080976BA300000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.032< 000000bd:86[17]:AEDD1BD080976BA3000000BD020100000500000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.004> 000000bd:00:0008A0D7D68BD0300A4700000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+1.001< (timeout)
PASS(test_InitSelfAborts())
t+0.003> 000000bd:81[99]:8AE6D9262721E9C172ABE7F2285606D73322A7B3B9125659EA2CE5BA5DEF01E7D5DA0EFCFBF7BE6DA2A55FCBFB66A22E8849E1415C379A4663
t+0.004> 000000bc:86[8]:7F00C06F02A844DC00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.028< 000000bc:bf[1]:060000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CHECK_EQ fail at test_InitOther[434]:r.init.cmd != U2FHID_INIT: (continuing -a)
CHECK_GE fail at test_InitOther[435]:MSG_LEN(r) < MSG_LEN(f2): (continuing -a)
[... 112 lines cut ...]
CHECK_EQ fail at test_Interleave[386]:isError(r, ERR_CHANNEL_BUSY) != true: (continuing -a)
t+0.012< 000000bd:00:A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7A7
t+1.001< (timeout)
CHECK_EQ fail at test_Interleave[392]:0 != U2Fob_receiveHidFrame(device, &r, 1.0): (continuing -a)
CHECK_EQ fail at test_Interleave[393]:r.cid != cid0: (continuing -a)
CHECK_EQ fail at test_Interleave[394]:r.cont.data[1] != expected: (continuing -a)
t+0.501< (timeout)
PASS(test_Interleave())
t+0.002> 00000100:81[10]:5E98B364C3DCBB9DB7570000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< 00000100:bf[1]:0B0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
CHECK_EQ fail at test_LeadingZero[460]:r.init.cmd != U2FHID_PING: (continuing -a)
CHECK_EQ fail at test_LeadingZero[461]:MSG_LEN(f) != MSG_LEN(r): (continuing -a)
PASS(test_LeadingZero())
t+2.002< (timeout)
PASS(test_Idle(2.0))
t+0.002> 00000000:86[8]:00FA37C6F832E5BE00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< 00000000:bf[1]:0B0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
PASS(test_NothingOnChannel0())
t+0.004> ffffffff:81[8]:411797343DCE73B800000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
t+0.024< ffffffff:bf[1]:0B0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
PASS(test_OnlyInitOnBroadcast())
test_Descriptor[617]: DESCRIPTOR: 06D0F10901A1010920150026FF007508954081020921150026FF00750895409102C0
PASS(test_Descriptor())

```

liamjm, 2021-11-02:
Digging into the first failure (test_InitOnNonBroadcastEchoesCID: https://github.com/google/u2f-ref-code/blob/master/u2f-tests/HID/HIDTest.cc#L464)....

In this test, an INIT command with non-broadcast CID is sent. The expectation is that the configured CID is returned. Instead an error is returned (CtapHid::ERR_INVALID_CHANNEL). 
This is "working as implemented":
a) The error is returned [in process_hid_packet()](https://github.com/google/OpenSK/blob/5e682d9e176e936c22fcb963a708ffb0b47a33e6/src/ctap/hid/mod.rs#L172)
b) via [inspecting the command and CID](https://github.com/google/OpenSK/blob/5e682d9e176e936c22fcb963a708ffb0b47a33e6/src/ctap/hid/mod.rs#L339)
c) and [ensuring the CID has been allocated](https://github.com/google/OpenSK/blob/5e682d9e176e936c22fcb963a708ffb0b47a33e6/src/ctap/hid/mod.rs#L344)

However, from what I can tell from the specs, the behavior expected in this test, is not described in the [reference specifications](https://fidoalliance.org/specs/fido-u2f-v1.2-ps-20170411/fido-u2f-hid-protocol-v1.2-ps-20170411.html#u2fhid_init). The other non-OpenSK device I have passes this test.

So this raises several options:
a) the spec is missing requirements that are generally expected and commonly implemented
b) OpenSK should implement this behavior
c) The reference code should remove this test as it is not a requirement.

kaczmarczyck, 2021-11-03:
Thanks for creating this issue. OpenSK is tested with FIDO's new conformance tool, described [here](https://fidoalliance.org/certification/functional-certification/conformance/). Unfortunately, this tool is closed source. The FIDO Alliance GitHub deprecated the tool you have been testing with (see [About section](https://github.com/fido-alliance/google-u2f-ref-code)).

I agree that we should have more open source, up-to-date testing for U2F, HID and NFC. Ideally from our perspective, they should be integrated into our [Test Tool](https://github.com/google/CTAP2-test-tool). This is on our list of future work.

kaczmarczyck, 2021-11-10:
Closing as working as intended.

closed 2021-11-10 by kaczmarczyck