google/OpenSK #406: Reset Security Key in Nordic nRF52840-dongle?

Lostsite, 2021-11-10:
1. Is there any way to remove all the stored data inside the Security Key (after successfully installed) without using JTAG or other devices connected? (Basically, I just want to know Can I reset the Security Key in Windows or in the Browser? (I can't reset it in windows, basically, it ask me to push the pushbuttons twice on the device, and that won't work).

2. Is there any way to erase all the data in Nordic nRF52840-dongle and reset it into the Factory Mode (set the storage like the new device I get from the factory).

3. If 2. is possible, is there any way I can do it through USB connections (without JTAG) (would `--programmer=nordicdfu` order still work?)

Lostsite, 2021-11-10:
Nevermind, the issue kind already been discribed in issue #179

kaczmarczyck, 2021-11-10:
I'm reopening this for myself, there seems to be a problem with our documentation. No action required from you. Thanks for exploring OpenSK!

closed 2021-11-19 by kaczmarczyck