google/OpenSK #411: Easier DFU for MDK

kaczmarczyck, 2021-11-18:
In #408, we add documentation for the MDK. This should be easier, and the deploy script generally allows DFU for other boards than the dongle. Let's check if we break its bootloader before we fix and recommend that, though (see #268).

ia0, 2025-11-14:
Did you try pressing reset while plugging the dongle? I think pressing reset while the dongle is powered is something that must be supported by the application, and I don't think this is the case here.

hitsmaxft, 2025-11-14:
> Did you try pressing reset while plugging the dongle? I think pressing reset while the dongle is powered is something that must be supported by the application, and I don't think this is the case here.

well, it doesn't works,  i can see the hex start address is  0x1000 , which should not break the mbr for open dfu bootloader on the ebyte dongle. Does this firmeware still erasing  uicr ? 

I will use jlink to recovery the bootloader and try it again