google/OpenSK #437: Nordic nRF52840 Dongle flash error

Rcrafte22, 2022-03-07:
## Steps to Reproduce the Problem

While I was trying to flash OpenSK

`[root@localhost OpenSK]# ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
info: Updating rust toolchain to nightly-2020-06-10
info: syncing channel updates for 'nightly-2020-06-10-x86_64-unknown-linux-gnu'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
info: Building Tock OS for board nrf52840_dongle_dfu
Traceback (most recent call last):
  File "./deploy.py", line 990, in <module>
    main(main_parser.parse_args())
  File "./deploy.py", line 749, in main
    OpenSKInstaller(args).run()
  File "./deploy.py", line 630, in run
    self.build_tockos()
  File "./deploy.py", line 351, in build_tockos
    self.checked_command(["make"], cwd=props.path, env=env)
  File "./deploy.py", line 283, in checked_command
    cmd, stdout=stdout, timeout=None, check=True, env=env, cwd=cwd)
  File "/usr/lib64/python3.6/subprocess.py", line 423, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib64/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib64/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'make': 'make'`

kaczmarczyck, 2022-03-07:
Hi! Thanks for reporting. Wild guess, sorry if you tried that already: Is `make` installed? Depending on your distro, it could be something like:
```
sudo apt-get install build-essential
```

Rcrafte22, 2022-03-07:
> Hi! Thanks for reporting. Wild guess, sorry if you tried that already: Is `make` installed? Depending on your distro, it could be something like:
> 
> ```
> sudo apt-get install build-essential
> ```

Thank you, I forgot to check this.

closed 2022-03-07 by Rcrafte22