google/OpenSK #448: Cannot RESET through Windows 10/11 and Chrome setting

superskybird, 2022-03-25:
## Expected Behavior
FIDO2 security key can be RESET through
- Chrome Settings->Manage Security Keys
chrome://settings/securityKeys?search=security+key
- Windows 10/11 Sign-in Options->Security Key

## Actual Behavior
show error.
Even though replug and press button inside 10 seconds.

ia0, 2022-03-27:
Thanks for the bug report!

I could actually reproduce on both Windows and Linux (Chrome).

Here is the trace for Linux after the replug:
```
Received message: Message { cid: [ff, ff, ff, ff], cmd: Init, payload: [18, 5b, 26, be, 24, 67, 4e, a7] }
Sending message: Message { cid: [ff, ff, ff, ff], cmd: Init, payload: [18, 5b, 26, be, 24, 67, 4e, a7, 00, 00, 00, 01, 02, 01, 00, 00, 05] }
Received message: Message { cid: [00, 00, 00, 01], cmd: Cbor, payload: [04] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: Cbor, payload: [00, af, 01, 83, 66, 55, 32, 46, 5f, 56, 32, 68, 46, 49, 44, 4f, 5f, 32, 5f, 30, 6c, 46, 49, 44, 4f, 5f, 32, 5f, 31, 5f, 50, 52, 45, 02, 85, 6b, 68, 6d, 61, 63, 2d, 73, 65, 63, 72, 65, 74, 6b, 63, 72, 65, 64, 50, 72, 6f, 74, 65, 63, 74, 6c, 6d, 69, 6e, 50, 69, 6e, 4c, 65, 6e, 67, 74, 68, 68, 63, 72, 65, 64, 42, 6c, 6f, 62, 6c, 6c, 61, 72, 67, 65, 42, 6c, 6f, 62, 4b, 65, 79, 03, 50, 56, d2, a9, 52, 0d, fb, 48, 6d, 9b, 5f, fa, 2b, 0f, 35, dd, dd, 04, aa, 62, 72, 6b, f5, 62, 75, 70, f5, 68, 61, 6c, 77, 61, 79, 73, 55, 76, f4, 68, 63, 72, 65, 64, 4d, 67, 6d, 74, f5, 69, 61, 75, 74, 68, 6e, 72, 43, 66, 67, f5, 69, 63, 6c, 69, 65, 6e, 74, 50, 69, 6e, f4, 6a, 6c, 61, 72, 67, 65, 42, 6c, 6f, 62, 73, f5, 6e, 70, 69, 6e, 55, 76, 41, 75, 74, 68, 54, 6f, 6b, 65, 6e, f5, 6f, 73, 65, 74, 4d, 69, 6e, 50, 49, 4e, 4c, 65, 6e, 67, 74, 68, f5, 70, 6d, 61, 6b, 65, 43, 72, 65, 64, 55, 76, 4e, 6f, 74, 52, 71, 64, f5, 05, 19, 1d, b9, 06, 82, 02, 01, 08, 18, 70, 09, 81, 63, 75, 73, 62, 0a, 81, a2, 63, 61, 6c, 67, 26, 64, 74, 79, 70, 65, 6a, 70, 75, 62, 6c, 69, 63, 2d, 6b, 65, 79, 0b, 19, 08, 00, 0c, f4, 0d, 04, 0f, 18, 20, 10, 08, 14, 18, 96] }
Received message: Message { cid: [00, 00, 00, 01], cmd: Cbor, payload: [01, a6, 01, 58, 20, e3, b0, c4, 42, 98, fc, 1c, 14, 9a, fb, f4, c8, 99, 6f, b9, 24, 27, ae, 41, e4, 64, 9b, 93, 4c, a4, 95, 99, 1b, 78, 52, b8, 55, 02, a1, 62, 69, 64, 66, 2e, 64, 75, 6d, 6d, 79, 03, a2, 62, 69, 64, 41, 01, 64, 6e, 61, 6d, 65, 65, 64, 75, 6d, 6d, 79, 04, 81, a2, 63, 61, 6c, 67, 26, 64, 74, 79, 70, 65, 6a, 70, 75, 62, 6c, 69, 63, 2d, 6b, 65, 79, 08, 40, 09, 01] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: Cbor, payload: [35] }
Received message: Message { cid: [00, 00, 00, 01], cmd: Cbor, payload: [07] }
Sending message: Message { cid: [00, 00, 00, 01], cmd: Cbor, payload: [30] }
```

We'll have to investigate more.

jmichelp, 2022-03-27:
If all these commands are sent by Chrome as part of a reset process, their logic may be wrong if I remember the specification correctly.
Here is the sequence I read from the dump: 

1. Init command to get an HID channel
2. CBOR 0x04 (GetInfo command)
3. CBOR 0x01 (MakeCredential command) and we return `ERR_PIN_NOT_SET`.
4. CBOR 0x07 (Reset command). And we return `ERR_NOT_ALLOWED`

@kaczmarczyck could confirm this but Reset being destructive, it's protected by a 10 seconds timer after plugging the key. On top of this, any "use" of the key would disable the Reset command. As Chrome is sending us a MakeCredential command, this should fall under the "usage" commands and invalidate the permission to send us a Reset.

I have no idea why they would send a MakeCredential before a Reset.
For the reference, here is the decoding of the CBOR payload associated with the MakeCredential command:
```
{
  1: h'E3B0C44298FC1C149AFBF4C8996FB92427AE41E4649B934CA495991B7852B855',
  2: {"id": ".dummy"},
  3: {"id": h'01', "name": "dummy"},
  4: [{"alg": -7, "type": "public-key"}],
  8: h'',  # pinUvAuthParam
  9: 1     # pinUvAuthProtocol
}
```

kaczmarczyck, 2022-03-27:
Our implementing might be a bit too strict for the real world, it seems. This MakeCredential call is the deprecated CTAP 2.0 way to select an authenticator. Search for "zero length pinUvAuthParam" in the [specification](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-20210615.html#sctn-makeCred-authnr-alg). The note below is explaining it:

> Note: This is done for backwards compatibility with CTAP2.0 platforms in the case where multiple authenticators are attached to the platform and the platform wants to enforce [pinUvAuthToken](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-20210615.html#getinfo-pinuvauthtoken) feature semantics, but the user has to select which authenticator to get the [pinUvAuthToken](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-20210615.html#puatoken-pinuvauthtoken) from. CTAP2.1 platforms SHOULD use [§ 6.9 authenticatorSelection (0x0B)](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-20210615.html#authenticatorSelection).

A quick fix would be that MakeCredential does not disable Reset anymore. The specification allows that. Thanks @superskybird for catching and reporting!

kaczmarczyck, 2022-03-28:
The fix is merged. @superskybird Feel free to close this bug when you confirmed it works for you!

geofli, 2022-03-28:
Tested, it works well. Thanks.

superskybird, 2022-03-28:
Tested, it works well. Thanks.

closed 2022-03-28 by superskybird