google/OpenSK #454: Not working with google?

bubundas17, 2022-04-06:
I am using the stable branch without any modifications. 
I have successfully built and flashed it to an nrf52840 Dongle.
after flashing 
I have ran
`./tools/configure.py --certificate=crypto_data/opensk_cert.pem --private-key=crypto_data/opensk.key`
[image]
and the certificate seems to be installed. 
and the OpenSK device seems to be working on https://demo.yubico.com/playground
but when I tried using it with a google account, it is failing to add. 
[image]

bubundas17, 2022-04-06:
Related Issue: #389

geofli, 2022-04-07:
I changed 
pub const USE_BATCH_ATTESTATION: bool = false;
to 
pub const USE_BATCH_ATTESTATION: bool = true;
then I can use OpenSK to work with google account.
It seems google account can not accept self-attestation which is returned by OpenSK's authenticatorMakeCredential.

kaczmarczyck, 2022-04-07:
Thanks @bubundas17 and @geofli ! From an OpenSK perspective, everything is working as expected. I'll leave this open to track, as it should by default definitely work for a Google login.

Please note that setting `USE_BATCH_ATTESTATION` to `true` with a self-generated attestation key makes you uniquely identifiable across websites. A workaround would be to generate a new random batch certificate on every registration, but that would just be a more convoluted way to not use attestation.

geofli, 2022-04-07:
Yes, I saw the contents, thank you @kaczmarczyck 
`Usually, the attestation private key is shared between a batch of at least
100,000 security keys of the same model. If you build your own OpenSK, your
private key is unique to you. This makes you identifiable across registrations:
Two websites could collaborate to track if registrations were attested with the
same key material. If you use OpenSK beyond experimentation, please consider
carefully if you want to take this privacy risk.`
So U2F is left to use batch certificate because `it is required by some websites to use U2F`, right?
Now I think Google Account turns to use WebAuthN (CTAP2) instead of U2F, this issue happened.

Now If call WebAuthN from Chrome on Android, the platform/browser sends U2F command. But with Chrome on Windows and macOS, the platform/browser sends CTAP2 commands. Is there a way for RP to choose to send CTAP2 or U2F command through WebAuthN instead of platform/browser controls that?

bubundas17, 2022-04-07:
If it is not privacy friendly to use a unique attestation private key. Is it safe to use the crypto_data folder shared by FeitianTech? And is it safe to use a private key which is being shared in a public repository?

https://github.com/FeitianTech/OpenSK/releases

geofli, 2022-04-07:
The certificate and private key shared here https://github.com/FeitianTech/OpenSK/releases are different with the materials inside OpenSK hardwares sold publicly.
https://github.com/FeitianTech/OpenSK/releases was just regenerated to show how to use the GUI firmware update tool.

bubundas17, 2022-04-07:
Some off topic questions:
Will it be possible to backup and restore the storage partition without a j-Link and restore it later?

geofli, 2022-04-07:
AFAIK, it is impossible to do that unless you use a J-Link.

kaczmarczyck, 2022-04-07:
@bubundas17 Can you tell me more about your use case? It has traditionally been considered a feature that secrets can not leave your security key. I believe there may be good reasons to want such a feature, and I'm curious about your ideas!

kaczmarczyck, 2022-04-07:
Users should have the choice whether they want to lock down access to JTAG or not. You can [configure OpenSK](https://github.com/google/OpenSK/blob/f2496a8e6d71a4e838884996a1c9b62121f87df2/tools/configure.py#L187) to be locked down for improved security. I'm not saying that this closes all attack vectors for a local attacker, but arguably security key users have advantages from inaccessible private key storage.

I'd suggest that you don't tinker with the device that holds your actual important secret. I do see the benefit of backing up your security key, but how to do that correctly and securely is a hard question. There is no standardized solution to this problem so far. Of course, since OpenSK is open source, you could add that functionality for yourself. :)

bubundas17, 2022-04-07:
I am already using a Yubico Key for daily use. 
If I consider adding this feature and create a PR, will you guys accept it?

jmichelp, 2022-04-07:
If you use a vendor specific command (like we did for our `configure.py` script) to dump the storage, this should at least come with the step of validating the user's PIN to succeed.
Maybe the command should also take an ECC public key as a parameter to protect the storage so that OpenSK can encrypt it properly before sending it over USB.

Injecting the storage into another key has some side effects though as it creates a clone of an existing key. Not sure right now how to deal with this in a good way.

At the moment the only idea that comes to my mind would be that OpenSK creates internally a unique encryption key, derived from hardware constants that survive even a chip erase and that is guaranteed to be unique per chip. This encryption key could be used to seal the storage before dumping it and used again to decrypt it for injection, ensuring that the storage is still bound to the OpenSK device.
As the encryption key is unique per chip it guarantees such a command can't be abused to create clones. CryptoCell has such key baked in but we only partially support CC310 for the bootloader (SHA256 module, not encryption).

bubundas17, 2022-04-07:
@geofli 
Setting 
`const USE_BATCH_ATTESTATION: bool = true;`
Has another side effect. 
Yubico's WebAuthn demo, is no longer accepting the key. 
https://demo.yubico.com/webauthn-technical/registration
It says,
`Attestation certificate must use version 3!`
[image]

kaczmarczyck, 2022-04-07:
I opened a new issue #455 to track that. Seems like an independent problem!

geofli, 2022-04-08:
@bubundas17 , yes, the automatically generated certificate is not compliant with https://www.w3.org/TR/webauthn-1/#packed-attestation-cert-requirements , so Yubico demo server rejects it.
I just changed `const USE_BATCH_ATTESTATION: bool = true` to try to check that Google server does not accept self-attestation. If you want to really change OpenSK to have batch attestation, there is still some work to do.

kaczmarczyck, 2022-05-30:
The work mentioned by @geofli is tracked under #457.
We are looking into the problem of supporting none attestation. Looks like OpenSK is working as intended, but I will keep this issue open to track our behavior for Google logins.

Firemanpl, 2023-01-23:
same
[image]

Firemanpl, 2023-01-23:
Even I changed `const USE_BATCH_ATTESTATION: bool = true`

kaczmarczyck, 2023-02-06:
After compiling with `USE_BATCH_ATTESTATION` set to `true`, did you also configure OpenSK to set a certificate?

```shell
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
```

And if so, can you tell me what website you used, plus all steps I need to reproduce?

Firemanpl, 2023-02-11:
ok just i editing USE_BATCH_ATTESTATION = true in mod file:
[image]
and flashing. After flashing I trying add phisic key to google account:
[image]
and after enter the pin and clicked button on phisic key windows still please me enter the pin. I can do this indefinitely. 
loop
{
enter the pin-> click phisic button
} 
[image]

Firemanpl, 2023-02-11:
I forgot that i install with stable branch.

kaczmarczyck, 2023-02-13:
It works for me, but I see you use Windows, and you don't seem to pass the Windows dialog. Is your problem Google specific, and all other websites work?

Firemanpl, 2023-02-15:
yep

Firemanpl, 2023-02-15:
not working on windows 10 and 11

kaczmarczyck, 2023-02-16:
I can't reproduce. On both `stable` and `develop`, the following steps work on Windows 10:

1. Erase storage (for a clean reset).
2. Set batch attestation to true.
3. Compile and flash with deploy.py
4. Configure the certificate
5. Optional: Set up a PIN (I tried with and without)
6. Go into Google account settings, setup 2nd factor with security key.
7. Click through all Windows dialogs, until the security key is listed in my Google account.

kaczmarczyck, 2023-02-16:
I don't have an update on registering on Google without batch attestation.

kaczmarczyck, 2024-01-25:
Login with Google works now, tested with the `develop` branch, no batch attestation and Chrome 120.0.

closed 2024-01-25 by kaczmarczyck