google/OpenSK #455: Batch attestation fails on Yubico's WebAuthn demo

kaczmarczyck, 2022-04-07:
Reported by @bubundas17 in #454 [here](https://github.com/google/OpenSK/issues/454#issuecomment-1091919933).

Tested with `stable` and `develop`, setting
`const USE_BATCH_ATTESTATION: bool = true;`

> Yubico's WebAuthn demo, is no longer accepting the key.
https://demo.yubico.com/webauthn-technical/registration
It says,
`Attestation certificate must use version 3!`

geofli, 2022-04-08:
https://www.w3.org/TR/webauthn-1/#packed-attestation-cert-requirements

bubundas17, 2022-04-08:
@jmichelp 
I have edited `gen_key_meterials.sh` line 81-96
and added 

`    "${openssl}" req \
      -new \
      -key "${opensk_key}" \
      -out "${opensk_cert_name}.csr" \
      -subj "/C=US/O=OpenSK/OU=Authenticator Attestation/CN=OpenSK Hacker Edition"
    "${openssl}" x509 \
      -req \
      -days 3652 \
      -in "${opensk_cert_name}.csr" \
      -CA "${ca_cert_name}.pem" \
      -extfile "${openssl_ext_file}" \
      -CAkey "${ca_priv_key}" \
      -CAcreateserial \
      -outform pem \
      -out "${opensk_cert_name}.pem" \
      -sha256`
and added this lines in {openssl_ext_file}

`
basicConstraints=CA:FALSE
`

And generated new certs. 
After doing that, the generated certificate came out to be x509v3
with this, the  OpenSK device seems to be working on the Yubico demo website.

I still do not know how to go about the extension OIDs, and how to add them to the generated certificate.

bubundas17, 2022-04-08:
https://www.w3.org/TR/webauthn-1/#packed-attestation-cert-requirements
according to this
>If the related attestation root certificate is used for multiple authenticator models, the Extension OID 1.3.6.1.4.1.45724.1.1.4 (id-fido-gen-ce-aaguid) MUST be present, containing the AAGUID as a 16-byte OCTET STRING. The extension MUST NOT be marked as critical.

As the root certificate is being generated on the end user's computer as a part of the setup script, Is it okay to ignore Extension OID?

kaczmarczyck, 2022-04-11:
Fixed, #457 tracks certificate version progress.

closed 2022-04-11 by kaczmarczyck