google/OpenSK #464: Button does nothing on MDK USB Dongle

SaphireLattice, 2022-04-21:
## Expected Behavior
The "press me" blinking stops and I'm able to use at least some U2F demo/etc

## Actual Behavior
Nothing happens, the LED keeps blinking. I am also unable to re-enter DFU mode again

## Steps to Reproduce the Problem

1. `./deploy.py --board=nrf52840_mdk_dfu --opensk --programmer=none`
1. `python3 uf2conv.py -c -f 0xada52840 -o target/opensk.uf2 target/nrf52840_mdk_dfu_merged.hex`
1. Copied the `uf2` firmware over after having to update from older bootloader version

## Specifications

- Version: 1.0.0, f2496a8e6d71a4e838884996a1c9b62121f87df2
- Platform: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/

kaczmarczyck, 2022-04-22:
Hi!

I tried on both the stable and develop branch, following [the docs](https://github.com/google/OpenSK/blob/stable/docs/boards/nrf52840_mdk.md) like you have been, and both work as intended. With the press of a button, the blinking stops and registration finishes.

I'm not sure if I can help you at this point. A few ideas:

You were able to enter DFU mode (which includes a button touch), but now it seems all button functionality is gone? You could try other button functionality, i.e. anything in Chrome settings here: chrome://settings/securityKeys. Not all of them are supported by OpenSK though.

Or if you don't mind breaking the device and like fiddling with hardware, open the plastic cover and check the button.

SaphireLattice, 2022-04-22:
Oookay, just tried it in chrome. U2F/FIDO does not work, but WebAuthn/FIDO2 works fine (at least on demo sites)

Firefox however uh.. just keeps the token asking and when I press the button, it faintly stutters and continues to blink >_>

Greeeeeat... sigh. Thanks for the tip on that though! 

Weird the bootloader died, though

kaczmarczyck, 2022-04-22:
Firefox problems might be related to missing U2F support. Have you configured your device? That adds a certificate that U2F needs.
```
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
```

Not much I can say about the bootloader, but thanks for reporting!

closed 2023-01-12 by kaczmarczyck