google/OpenSK #475: OpenSK and ECDAA

geo-lit, 2022-05-01:
Hi I was in contact with @jmichelp asking him about ECDAA. 

Just wondering if anyone knows why the draft has not been implemented, and has been stagnant since 2018 (although I recently found a new draft from FIDO from May 2021 https://fidoalliance.org/specs/common-specs/fido-ecdaa-algorithm-v2.1-rd-20210525.pdf) I know that it's not in the specification, but seems like it would solve any privacy issues that come with batch attestation, which is not 100% private.

Also looking on guidance in terms of how I could start it's implementation using OpenSK. I tried familiarizing myself with the repo, but am a bit lost, so any guidance would be appreciated, specifically attestation flow, and where ECDAA would be added from the program-flow perspective.

kaczmarczyck, 2022-05-02:
I don't expect widespread usage of ECDAA. I think the underlying cryptography is relatively young, i.e. the 2021 draft fixes security holes from 2018. At the moment, I recomment to solve attestation related privacy issues by simply not using batch attestation. This is the default in OpenSK. It might fail for some websites with strict policies, but I don't expect OpenSK to be on their whitelists anyway. I'd argue that ECDAA's additional complexity is not worth the marginal benefit. Without attestation already, security keys are a better second factor than most.

That is for our product decision on ECDAA on OpenSK. However, from a research perspective, I welcome anyone to implement and experiment with it. A few thoughts and pointers:

1. In case of expected binary size increases, it's code should be behind a feature flag.
2. All cryptography can be added to our [software library](https://github.com/google/OpenSK/tree/develop/libraries/crypto/src). The EC basics are already there, for ECDSA and ECDH.
3. Batch attestation, if enabled, is set in [`process_make_credential` here](https://github.com/google/OpenSK/blob/a0e11bd5aa01bf73d9ccd4336524f96316962475/src/ctap/mod.rs#L859). It should give you some ideas about how to replace it with a different mechanism.

Feel free to tell me more about where exactly you're lost or ask more concrete questions, then I can give you a better answer where to look.

kaczmarczyck, 2023-11-07:
There are no new developments around ECDAA, so I assume no interest. Please reopen when this changes.

closed 2023-11-07 by kaczmarczyck