google/OpenSK #485: Implementing OpenSK-based FIDO2 emulator for QEMU

egor-duda, 2022-05-29:
I've made a quick-and-dirty prototype of virtual FIDO2 authenticator for QEMU using OpenSK application. The purpose is similar to #288, but instead of creating a CTAP library and linking it into QEMU, I've patched QEMU to add a fido2 device, which uses socket-connected chardev. Then I've modified main OpenSK application to listen on a unix socket and, after connection from QEMU, to process all CTAP requests/responses via that socket connection.

To go beyond prototype, I suppose I have to:

1. Create separate NonEmbeddedEnv (naming suggestions are welcome) for running on non-embedded OS
2. Implement file-based storage for persistent authenticator data. Currently I've used TestEnv with its BufferStorage, which means that master key is being lost after OpenSK application restart, making all previously created credentials invalid
3. Move main protocol I/O function from loop in src/main.rs to Env, and implement socket- or stdio- based I/O in NonEmbeddedEnv
4. Implement user-presence checking in new NonEmbeddedEnv

Obviously, all this new NonEmbeddedEnv stuff should not be compiled in when building token-targeted version of an application.

Is this a viable plan?

egor-duda, 2022-06-01:
> May I ask if you have a specific goal in mind? Is QEMU the purpose like in #288, or do you want a desktop OpenSK?

QEMU, mostly. 

I've experimented with running desktop version of OpenSK (on Linux, using /dev/uhid), but I don't see the point, security-wise, of running authenticator on the same host system with user agent. Maybe only for testing/development?

[...]
> All of this is tracked under a [project](https://github.com/google/OpenSK/projects/7).

Great!

I've implemented storage which stores its contents in host-OS-based file: https://github.com/egor-duda/OpenSK/tree/hostenv-file-storage
This change is independent of Env-related changes -- shall I make a PR from it?

[...]

> Since you were asking for "beyond prototype", I assume you mean that you want to create a PR that we would eventually merge. Cleanest and easiest would be for us to first finish the CTAP library, and then merging your new `NonEmbeddedEnv`. Else we'd have to port yet another platform to the new API. Afterwards, you could use that desktop version of OpenSK for whatever you want, for example QEMU. Is this a helpful answer for what you have in mind?

Absolutely, thanks for detailed response!

egor-duda, 2022-06-01:
I'm not sure if it's possible to make pure file-based storage without memory buffer using current API?

`read_slice` returns a reference, which means, as far as I understand, than underlying storage object must own all those slices' data, and keep it while application is running.

ia0, 2022-06-01:
I see. That's actually a good reason to change the API then. I was considering it for other reasons (in particular make it compatible with `embedded-storage`), but this one adds to it. Might be worth checking whether it's easy to do and if it changing the binary size in any way.

egor-duda, 2022-06-02:
> Otherwise the storage end up being nothing more than a cookie jar that could be stolen by a malware while the promise of FIDO is to resist a remote attacker.

The idea is to have file-backed storage residing on the host OS, on which QEMU is running, whereas user agent runs on the guest, accessing authenticator via virtual QEMU device. Presumably, malware shouldn't be able to escape from guest VM and access any data on host OS.

If malware somehow succeeds in reaching host OS, then I don't think file encryption would be of much help. If malware gets non-root access to host OS, then normal OS permission protection should be enough. If malware escapes from guest and gets root-level access to host OS, then it would be able to get decrypted storage from memory of running authenticator application.

kaczmarczyck, 2022-06-03:
@ia0 is looking into solutions to enable file storage (#486, #487). Outside of that, any other plans regarding implementation for this issue? Feel free to [reach out](mailto:opensk@google.com) to discuss if you are interested!

kaczmarczyck, 2022-06-07:
Some thoughts on transports in FIDO. Generally, the WebAuthn transport we would use is this [internal](https://www.w3.org/TR/webauthn/#dom-authenticatortransport-internal). However, [CTAP only mentions HID, NFC and BLE](https://fidoalliance.org/specs/fido-v2.1-ps-20210615/fido-client-to-authenticator-protocol-v2.1-ps-20210615.html#fido-interfaces). Since WebAuthn allows it, I believe we should be okay to use it regardless. The intention I read into the specification is: If you don't use special hardware, implement it any way you like.

If you want the security key to look like a HID (or other) device, we might have to approach that differently. The question is at what level you want to exchange data.

kaczmarczyck, 2023-11-07:
Closing this as a duplicate of #288 .

closed 2023-11-07 by kaczmarczyck