google/OpenSK #507: cargo_clippy github workflow fails on forked repo

egor-duda, 2022-06-23:
After pushing changes to fork repo, `cargo_clippy` workflow fails with error "Resource not accessible by integration", see e.g. https://github.com/egor-duda/OpenSK/actions/runs/2547812258
This also hides real clippy errors because next `run: cargo clippy ...` command is not run after `actions-rs/clippy-check` failure.

This is caused by actions-rs/clippy-check#2, due to inability of `clippy-check` to add annotations, while running with read-only token.

Adding check for running `clippy-check` action only in main repo fixes this issue, and starts reporting `cargo clippy` errors in forked repos
```
diff --git a/.github/workflows/cargo_clippy.yml b/.github/workflows/cargo_clippy.yml
index 7d10e1c..ac4e08d 100644
--- a/.github/workflows/cargo_clippy.yml
+++ b/.github/workflows/cargo_clippy.yml
@@ -23,6 +23,7 @@ jobs:
         run: ./setup.sh
 
       - uses: actions-rs/clippy-check@v1
+        if: github.repository == 'google/OpenSK'
         with:
           token: ${{ secrets.GITHUB_TOKEN }}
           args: --all-targets --features std

```

kaczmarczyck, 2022-06-27:
Thanks for reporting. I'll look into it when I have time. Seems to be not too high impact, since you can always test locally with `run_desktop_tests` and it works on our main repository.

kaczmarczyck, 2023-11-09:
The script should now run clippy, locally or in the repository of your choice. Please reopen if the problem persists.

closed 2023-11-09 by kaczmarczyck