google/OpenSK #519: User presence touch timeouts get treated as user presence

liamjm, 2022-07-28:
## Expected Behavior
When waiting for user presence, a timeout should lead to an error, eg CTAP2_ERR_USER_ACTION_TIMEOUT

## Actual Behavior
Successful response is returned to host.

## Steps to Reproduce the Problem

1.  send a AuthenticatorMakeCredential command
1.  don't touch the buttons

## Specifications

- Version: develop branch
- Platform: linux

liamjm, 2022-07-28:
Issue is caused by the result variable from the presence check being overwritten by the result from sending the keepalive packet:

https://github.com/google/OpenSK/blob/develop/src/ctap/mod.rs#L352-L355

```
result = env.user_presence().wait_with_timeout(...);
if !matches!(result, Err(UserPresenceError::Timeout)) {
   break;
 }   
result = send_keepalive_up_needed(env, channel, KEEPALIVE_DELAY);
if result.is_err() {
  break;
}
```

kaczmarczyck, 2022-08-02:
I checked on the latest develop. If I wait for touch timeout, it returns `CTAP2_ERR_USER_ACTION_TIMEOUT` as expected. Can be closed?

liamjm, 2022-08-08:
Yep, SGTM.

PR #521 (2022-08-02): Fix user presence by not overwriting error with OK()
Fixes #519 

The key change is in check_user_presence(), where the result of the user_presence().wait_with_timeout() call is now not overwritten unless the subsequent call to send_keepalive_up_needed() returns an error.

Tests added in #520 demonstrate the bug.
closed 2022-08-08 by liamjm