google/OpenSK #544: LED flashing light blue after flashing, cannot enter DFU mode

Tiebe, 2022-09-18:
## Expected Behavior
Being able to use OpenSK after flashing

## Actual Behavior
Led flashing light blue rapidly, cannot enter DFU mode by pressing the button while plugging in.

lsusb output:
```
Bus 003 Device 053: ID 1915:521f Nordic Semiconductor ASA OpenSK
```

dmesg output:
```
[ 4676.899717] usb 3-1: new full-speed USB device number 53 using xhci_hcd
[ 4677.081222] usb 3-1: New USB device found, idVendor=1915, idProduct=521f, bcdDevice= 0.01
[ 4677.081233] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4677.081238] usb 3-1: Product: OpenSK
[ 4677.081242] usb 3-1: Manufacturer: Nordic Semiconductor ASA
[ 4677.081245] usb 3-1: SerialNumber: v1.0
[ 4677.085530] hid-generic 0003:1915:521F.0031: hiddev96,hidraw6: USB HID v1.10 Device [Nordic Semiconductor ASA OpenSK] on usb-0000:00:14.0-1/input0
```

## Steps to Reproduce the Problem

1. `git clone https://github.com/google/OpenSK.git`
2. `curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs/ | sh`
3. `source $HOME/ .cargo/env`
4. `cd OpenSK`
5. `./setup.sh`
6. `./deploy.py --board=nrf52840_mdk_dfu --opensk --programmer=none`
7. `python3 uf2conv.py -c -f 0xada52840 -o target/opensk.uf2 target/nrf52840_mdk_dfu_merged.hex`
8. Copy opensk.uf2 to usb dongle

## Specifications

- Version: `[a79abc2](https://github.com/google/OpenSK/commit/a79abc209a8ce15d72274531c69f59fe72f86e88)`
- Platform: nRF52840-MDK

kaczmarczyck, 2022-09-20:
Works for me on a fresh MDK. Maybe you need to erase the persistent storage, and the blinking is a panic. In that case, follow the instructions above again, but instead of flashing OpenSK, you install the example app that erases all storage.
WARNING: Erases all existing credentials, in case you have any important ones.

```
./deploy.py --board=nrf52840_mdk_dfu --programmer=none --erase_storage
python3 uf2conv.py -c -f 0xada52840 -o target/opensk.uf2 target/nrf52840_mdk_dfu_merged.hex
```

Then copy and replug. Give it a few seconds to complete erasing the storage. Then try again with `--opensk` instead of `--erase_storage`:

```
./deploy.py --board=nrf52840_mdk_dfu --programmer=none --opensk
python3 uf2conv.py -c -f 0xada52840 -o target/opensk.uf2 target/nrf52840_mdk_dfu_merged.hex
```

Replug and please report back if that changes anything.

Tiebe, 2022-10-01:
Would love to try this, but I can't even enter dfu mode, so I can't flash anything

JunASAKA, 2022-10-02:
> Would love to try this, but I can't even enter dfu mode, so I can't flash anything

I think it is impossible to be failed to enter DFU mode. You can try pressing the button and reset key(backside installed one) at the same time when the device is already connected to the PC. Then it will go into DFU if nothing went wrong. 
And you can then flash it with
‘./deploy.py --board=nrf52840_dongle_dfu --erase_storage --programmer=nordicdfu’

‘./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu’

If it surely cannot go into DFU, you can try to flash it with J-Link or other programmers.

kaczmarczyck, 2022-10-03:
Ah right, understood. Is this problem OpenSK related, because you can't enter DFU mode since you flashed an OpenSK, and it worked before?

Tiebe, 2022-10-03:
That's correct

kaczmarczyck, 2022-10-03:
In that case, I'll have to understand how to reproduce. Please let me know if you have any pointers etc. about that.

chimpnz, 2022-10-17:
I had the same problem. Flashing LED and unresponsive bootloader. 

If you have a Raspberry Pi you can use OpenOCD to re-flash the UF2 bootloader.

I followed the instructions here https://www.rototron.info/circuitpython-nrf52840-dongle-openocd-pi-tutorial/

It is for the PCA10059 dongle but the process is the same. The only difference is the location of the SWDCLK and SWDIO pins. The pinout can be found here https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/#pinout-diagram

Heed the warnings about resetting the voltage regulator.

The latest 0.7.0 UF2 bootloader for the MDK is here:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/download/0.7.0/mdk_nrf52840_dongle_bootloader-0.7.0_s140_6.1.1.hex

Only flash the bootloader and skip installing CircuitPython at the end.

OpenSK ran fine with the fresh bootloader.

kaczmarczyck, 2022-10-17:
Thanks for helping! Do you know if the unresponsive bootloader was caused by OpenSK? @chimpnz

chimpnz, 2022-10-17:
> Thanks for helping! Do you know if the unresponsive bootloader was caused by OpenSK? @chimpnz

Not really sure. I tried to reproduce it and can not. The MDK dongle that had the issue had CircuitPython and LOGItacker installed previously. I tried again loading both via UF2 and then OpenSK and the bootloader remains functional.

I tried another MDK dongle with the original bootloader (UF2 Bootloader 0.2.13-42-g82464f9-dirty) with CircuitPython/LOGItacker/OpenSK and bootloader works fine.

One odd thing is that CircuitPython will no longer run on either dongle. I can copy the UF2 file but it does not run after reset. Tried several versions. LOGItacker and OpenSK run fine.

jmichelp, 2022-10-17:
Can you confirm which branch you're using? `stable` or `develop`?
The original issue is against `stable`.

Considering the dongle has been used before, this is what may have happened:
* other apps have configured the `UICR` NVM registers a specific way for their usage
* OpenSK at boot up checks that the reset pin and the voltage regulator (stored in `UICR`) are configured correctly [code](https://github.com/google/OpenSK/blob/stable/boards/nordic/nrf52840_mdk_dfu/src/main.rs#L212-L217)
* if not it wipes out the `UICR` values to put them in a known state
* this has the side effect of erasing the values used by the DFU/U2F bootloader

In `develop` branch we removed that part of the code to avoid such issue and we also have a [patch](https://github.com/google/OpenSK/blob/stable/patches/tock/04-update-uicr.patch) to protect these 2 bootloader values.

But the `develop` branch is not always tested with the MDK dongle so it might not work out of the box if we haven't properly kept the board configuration.

If you're using the `stable` branch, we can try backporting the patch to see if this solves the issue.

chimpnz, 2022-10-17:
I am using the stable branch.

PR #560 (2022-10-18): Fix nRF52 boards with bootloader
Backporting patch from develop branch which saves and restores the bootloader parameters in case we need to erase UICR at startup.

Fixes #544
closed 2022-10-18 by jmichelp