google/OpenSK #545: Request support E104-BT5040U （nRF 52840）

yazhuodong-wyze, 2022-09-19:
product link : https://www.ebyte.com/product-view-news.html?id=1030
buy link : https://detail.tmall.com/item.htm?id=624428606924

kaczmarczyck, 2022-09-20:
Hi! Can you please specify what kind of support you request? If you want to run OpenSK on your hardware, the Nordic chip is supported. However, BLE is not yet implemented, so you'd have to contribute for that :)

yazhuodong-wyze, 2022-09-20:
Hello Again.  I need an installation document, like this: https://github.com/google/OpenSK/blob/stable/docs/boards/nrf52840_mdk.md
, no need for Bluetooth for now.I mentioned E104-BT5040U, this one. I think it uses an nRF 52840.

kaczmarczyck, 2022-09-20:
We don't support new hardware on request. We do accept contributions, so if you find out how to install OpenSK on this platform, feel free to extend our documentation.

yazhuodong-wyze, 2022-09-20:
ok，thanks.I will try this new board, but I am not good at embedded development. If it works, I will provide documentation.

chimpnz, 2022-10-26:
I have 3 of these. It comes with a broken Open DFU bootloader.
When first flashed with the OpenSK for the PCA10059 via nrfutil or nrf Connect it becomes completely unresponsive.

The bootloader for the PCA10059 found [here](https://devzone.nordicsemi.com/f/nordic-q-a/40924/how-can-i-restore-the-original-bootloader-of-a-pca10059) will restore functionality. With the PCA10059 bootloader it behaves just like a PCA10059. The LEDs and buttons are connected to the same io pins. The left button is RESET and the right USER.

[image]

I would not recommend these. For the same price you can get a PCA10059 with a working bootloader. The plastic case it comes with is also not very useful. The buttons are not accessible. Also one leg of the RESET button was not soldered on one board and the USER button was not soldered on another. 

[image]

Jiffies-Duke, 2022-10-29:
Same me. I flashed the firmware into a BT5040U. It then disconnected automatically, could not be found by PC anymore.
After flashed a bootloader from nRF5 SDK by using SWDIO interface, it came back and functioned right.
Bootloader I used: https://devzone.nordicsemi.com/cfs-file/__key/communityserver-blogs-components-weblogfiles/00-00-00-00-13/pca10059_5F00_bootloader.zip
Programmer hardware: DAPLink or CMSIS-DAP
Programmer software: pyocd
By the way, BT5040U in China is much cheaper than PCA10059, nearly half the price.

Firemanpl, 2023-01-06:
On aliexpress. It board seems to be the cheapest board with nrf52840 form me (at least form my country e.g. Poland). I have one question. @chimpnz How you flashed this bootloader. jlink, Daplink or something other maybe ? Because I maked this same, and after i flashed i saw this issue.

chimpnz, 2023-01-06:
> On aliexpress. It board seems to be the cheapest board with nrf52840 form me (at least form my country e.g. Poland). I have one question. @chimpnz How you flashed this bootloader. jlink, Daplink or something other maybe ? Because I maked this same, and after i flashed i saw this issue.

I used a OpenOCD on a Raspberry Pi. Mostly followed the directions here https://www.rototron.info/circuitpython-nrf52840-dongle-openocd-pi-tutorial 

Just substitute to correct SWD pins on the different device and flash the bootloader from https://devzone.nordicsemi.com/f/nordic-q-a/40924/how-can-i-restore-the-original-bootloader-of-a-pca10059

kaczmarczyck, 2023-01-09:
Thanks at @Firemanpl and @chimpnz for your extensive documentation here! About missing access to the button: If you want to port OpenSK to this board, how would you solve this? If you want to build a touchless security key, I think you have 2 options:

1. Somewhat conform to FIDO and use replugging for user presence checks. Not very user friendly, and means that the Reset command isn't as protected as before.
2. Pretend that the button was touched when asked. Loses the security added by requiring touch for sensitive actions, and probably doesn't conform to FIDO. Likely still better than most password based solutions, but more for personal use.

Or is there something entirely different you are trying to do?

zalexua, 2023-09-06:
Let me post some details here.

I've disappointment to select another model - E104-BT5040UA, as description was not clear enough that it's very different from E104-BT5040U. 
This works as a snuffer only, and you cannot use it with "nRF Connect at" all. You can use as sniffer in Wireshark only! 

[image]
[image]

[image]
[image]

Description on other sale lots (which I've missed) says:
```
Differences between E104-BT5040U and E104-BT5040UA:
1. The factory software of the two modules is different, and the functions are also different.
E104-BT5040U:  The main function is to support Nordic's official PC-side nrf connect, customers can use this module with nrf connect for BLE function debugging and development;
E104-BT5040UA: The main function is to capture BLE4.2 and BLE5.0 protocol packets, and it is used with other BLE modules to capture and analyze the communication of other modules.

2. The hardware structure is different.
E104-BT5040U:  It is developed based on Nordic's official nRF52840 USB Dongle module, the hardware structure is similar to it, and most of the IO ports and hardware resources of the chip are drawn;
E104-BT5040UA: It is developed based on Nordic's official development board PCA10056. In addition to the nRF52840 chip itself, a serial port chip is added, and only the programming interface is introduced. IO and other hardware resources have no interface.

3.E104-BT5040U is compatible with all the programs of Nordic's original nRF52840 USB Dongle, and IO ports and hardware resources have been introduced, which is convenient for customers to carry out secondary development;
E104-BT5040UA does not recommend customers to carry out secondary development, and Ebyte does not provide secondary development support.
```