google/OpenSK #549: Cargo Error: "error[E0635]: unknown feature `proc_macro_span_shrink`" when building.

JunASAKA, 2022-09-25:
## Actual Behavior
　　When running deploy script(develop branch) to build openSK or erase_storage for nrf52840_dongle with "./deploy.py --board=nrf52840_dongle_dfu --erase_storage --programmer=nordicdfu", it goes wrong showing "error[E0635]: unknown feature 'proc_macro_span_shrink'". More info shows below:
<br />

`info: Building example erase_storage`
`   Compiling proc-macro2 v1.0.44`
`   Compiling num-iter v0.1.43`
`   Compiling num-rational v0.3.2`
`error[E0635]: unknown feature 'proc_macro_span_shrink'`
`  --> /home/herrscherzero/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.44/src/lib.rs:92:30`
`   |`
`92 |     feature(proc_macro_span, proc_macro_span_shrink)`
`   |                              ^^^^^^^^^^^^^^^^^^^^^^`
``
`error: aborting due to previous error`
``
`For more information about this error, try 'rustc --explain E0635'.`
`error: could not compile 'proc-macro2'`
``
`To learn more, run the command again with --verbose.`
`warning: build failed, waiting for other jobs to finish...`
`error: build failed`
`fatal: Failed to execute cargo: Command '['cargo', 'build', '--release', '--target=thumbv7em-none-eabi', '--features=with_ctap1', '--example', 'erase_storage']' returned non-zero exit status 101.`
``
## Specifications

- Version: f2fac83
- Platform: Arch Linux (x86_64)

JunASAKA, 2022-09-26:
> Initial assessment: I can reproduce after a `cargo update`. It did a suspicious version jump there: `proc-macro2 v1.0.43 -> v1.0.44` So before we have it fixed, I suspect you can lock the crate's version.

Thanks for replying.
I noticed that the new released (commit for proc-macro2)[https://github.com/dtolnay/proc-macro2/commit/cc97264f9f324ade9b820a1629f379b2cbb6e632] showes that a new feature "proc_macro_span_shrink" was added to src/lib.rc. After removing such feature, the compilation would succeed.
But it is shamed that I haven't got alone with Rust, Any suggestion with that? Looking forward to your early reply.

Thanks and Regards,
Jun.

JunASAKA, 2022-09-27:
The issued has been fixed.
Closed.

closed 2022-09-27 by JunASAKA

johnvsnagendra, 2022-10-10:
Hi, I have the same error on Mac M1 and tried adding `proc-macro2 = "=1.0.43"` in dependencies. But it did not work. 

In return it gave me a bunch of errors that look like these. Am I missing something or Is there any other fix?
```bash
error[E0433]: failed to resolve: could not find `Span` in `proc_macro`
  --> /Users/johnversus/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.43/src/detection.rs:67:49
   |
67 |     let works = panic::catch_unwind(proc_macro::Span::call_site).is_ok();
   |                                                 ^^^^ could not find `Span` in `proc_macro`

error[E0433]: failed to resolve: could not find `Spacing` in `proc_macro`
   --> /Users/johnversus/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.43/src/wrapper.rs:158:47
    |
158 |                 Spacing::Joint => proc_macro::Spacing::Joint,
    |                                               ^^^^^^^ could not find `Spacing` in `proc_macro`

error[E0433]: failed to resolve: could not find `Spacing` in `proc_macro`
   --> /Users/johnversus/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.43/src/wrapper.rs:159:47
    |
159 |                 Spacing::Alone => proc_macro::Spacing::Alone,
    |                                               ^^^^^^^ could not find `Spacing` in `proc_macro`

error[E0433]: failed to resolve: could not find `token_stream` in `proc_macro`
   --> /Users/johnversus/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.43/src/wrapper.rs:303:26
    |
303 |     Compiler(proc_macro::token_stream::IntoIter),
    |                          ^^^^^^^^^^^^ could not find `token_stream` in `proc_macro`

```

kaczmarczyck, 2022-10-10:
We did not backport the fix to stable yet. Can you confirm that you are on the stable branch? In that case, I'll respond on #553 since it seems to be same problem.

johnvsnagendra, 2022-10-10:
Yes, I am on stable and I tested on the `nightly` and `span-locations` features with the same version, and it still throws an error.

kaczmarczyck, 2022-10-10:
Please let me know if your problem is solved on stable now!

johnvsnagendra, 2022-10-12:
Hey @kaczmarczyck, yup it is solved now. Thanks 🤩

PR #550 (2022-09-27): Fixed proc-macro2 version
Fixes #549

Pins the version for our proc-macro2 dependency.