google/OpenSK #553: proc-macro2 compilation error

Samthebest999, 2022-10-09:
## Expected Behavior
**Compiling** proc-macro2 v1.0.46
## Actual Behavior
error[E0635]: unknown feature `proc_macro_span_shrink`
  --> /home/{username}/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.46/src/lib.rs:92:30
   |
92 |     feature(proc_macro_span, proc_macro_span_shrink)
   |                              ^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0635`.
error: could not compile `proc-macro2`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
fatal: Failed to execute cargo: Command '['cargo', 'build', '--release', '--target=thumbv7em-none-eabi', '--features=with_ctap1']' returned non-zero exit status 101.

## Steps to Reproduce the Problem

git clone https://github.com/google/OpenSK.git
cd OpenSK
pip3 install nrfutil
./setup.sh
sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/
sudo udevadm control --reload
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
## Specifications
Fetian OpenSK Dongle V2
- Version:
The current one at the time of posting. (Sorry I don't know)
- Platform:
Ubuntu 22.04.1 LTS Kernel: Linux 5.15.0-48-generic x86_64

kaczmarczyck, 2022-10-10:
Can you pull and see if it works?

Samthebest999, 2022-10-10:
It works now! Thanks!

PR #555 (2022-10-10): Submit Cargo.lock
Fixes #553. Should also fix the new issue mentioned on #549.

This is a backport in the spirit of #550. Please double check if your Rust installation behaves the same as mine.