google/OpenSK #554: Manual building for unsupported boards (simulation)

L0g4n, 2022-10-09:
Hello there,

I want to (try to) build OpenSK for testing purposes for other (experimental) boards which are supported by Tock OS, however, not supported by this project. In my case specifically the OpenTitan Verilator Simulation.

However, after checking your readme, I can only, understandably, see instructions for calling the `deploy.py` script for supported boards such as the nRF52840. What would be the workflow for manually triggering the building process and handling the flashing process?

kaczmarczyck, 2022-10-10:
If your board is supported by Tock, then do you know how to install it? Our flow is roughly this:

- `make` our custom TockOS
- build the Rust app, flags like in [this function](https://github.com/google/OpenSK/blob/a79abc209a8ce15d72274531c69f59fe72f86e88/deploy.py#L361)
- flash Tock with a method of your choice
- use Tockloader to write the app

Flashing generally depends on your hardware. For inspiration, we describe different methods in our `docs/`, e.g. DFU, JTAG, OpenOCD. Otherwise, if you look for `checked_command` in our deploy script, you will see the commands we call.

Is that roughly what you wanted to know? I can add more detail if you have any concrete problems?

jmichelp, 2022-10-10:
In addition, pay attention that a board supported by Tock is unfortunately not enough as we have patches being applied to the kernel to support both our flash storage and a USB capsule which we rely on.
Those patches were not accepted upstream and therefore we need to maintain them in our repository.
Last but not least, we're currently based on Tock 1.x (more precisely the commit right before Tock moved to 2.0).

closed 2022-11-11 by L0g4n