google/OpenSK #562: Testing invariants fails with "internal compiler error" on nightly-2021-03-25 toolchain

L0g4n, 2022-10-21:
## Expected Behavior

Testing the invariants during the deploy script should pass.

## Actual Behavior

The tests fail.

## Steps to Reproduce the Problem

- Run the deploy script or manually run `cargo test --features=std --lib customization` and watch the output.
- `./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu`
 
```
info: Updating rust toolchain to nightly-2021-03-25
info: syncing channel updates for 'nightly-2021-03-25-x86_64-unknown-linux-gnu'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
Done: The submodules and patches are in sync.
info: Building Tock OS for board nrf52840_dongle_dfu
   Compiling tock-tbf v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/libraries/tock-tbf)
   Compiling tock-registers v0.6.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/libraries/tock-register-interface)
   Compiling tock-cells v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/libraries/tock-cells)
   Compiling enum_primitive v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/libraries/enum_primitive)
   Compiling tock-rt0 v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/libraries/tock-rt0)
   Compiling tickv v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/libraries/tickv)
   Compiling nrf52840_dongle_dfu v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/boards/nordic/nrf52840_dongle_dfu)
   Compiling kernel v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/kernel)
   Compiling cortexm v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/arch/cortex-m)
   Compiling nrf5x v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/chips/nrf5x)
   Compiling capsules v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/capsules)
   Compiling cortexm4 v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/arch/cortex-m4)
   Compiling nrf52 v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/chips/nrf52)
   Compiling nrf52840 v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/chips/nrf52840)
   Compiling components v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/boards/components)
   Compiling nrf52_components v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/tock/boards/nordic/nrf52_components)
    Finished release [optimized + debuginfo] target(s) in 13.59s
info: Building OpenSK application
Testing invariants in customization.rs...
   Compiling libtock_core v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/libtock-rs/core)
   Compiling sk-cbor v0.1.2 (/home/yk/Dokumente/Coding/opensk_fork/libraries/cbor)
   Compiling persistent_store v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/libraries/persistent_store)
   Compiling libtock_codegen v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/libtock-rs/codegen)
   Compiling ctap2 v1.0.0 (/home/yk/Dokumente/Coding/opensk_fork)
   Compiling libtock_drivers v0.1.0 (/home/yk/Dokumente/Coding/opensk_fork/third_party/libtock-drivers)
thread 'rustc' panicked at 'found unstable fingerprints for item_children(libtock_codegen[2f0e]): [Export { ident: make_read_env_var#0, res: Def(Macro(Bang), DefId(18:15 ~ libtock_codegen[2f0e]::make_read_env_var)), span: /home/yk/Dokumente/Coding/opensk_fork/third_party/libtock-rs/codegen/src/lib.rs:24:1: 24:60 (#0), vis: Public }, Export { ident: main#0, res: Def(Macro(Attr), DefId(18:16 ~ libtock_codegen[2f0e]::main)), span: /home/yk/Dokumente/Coding/opensk_fork/third_party/libtock-rs/codegen/src/lib.rs:75:1: 75:63 (#0), vis: Public }]', /rustc/07e0e2ec268c140e607e1ac7f49f145612d0f597/compiler/rustc_query_system/src/query/plumbing.rs:593:5
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

error: internal compiler error: unexpected panic

note: the compiler unexpectedly panicked. this is a bug.

note: we would appreciate a bug report: https://github.com/rust-lang/rust/issues/new?labels=C-bug%2C+I-ICE%2C+T-compiler&template=ice.md

note: rustc 1.53.0-nightly (07e0e2ec2 2021-03-24) running on x86_64-unknown-linux-gnu

note: compiler flags: -C embed-bitcode=no -C debuginfo=2 -C incremental --crate-type lib

note: some of the compiler flags provided by cargo are hidden

query stack during panic:
#0 [item_children] collecting child items of `libtock_codegen`
#1 [visible_parent_map] calculating the visible parent map
end of query stack
error: could not compile `libtock_core`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
fatal: Failed to execute cargo: Command '['cargo', 'test', '--features=std,with_ctap1', '--lib', 'customization']' returned non-zero exit status 101.
```

## Specifications

- Version: 44cafb956612c343470649f93bd7f6bda9640f49 of `develop` branch

I'm not actually sure if I should report this here since it clearly says it's an internal compiler bug.

kaczmarczyck, 2022-10-21:
Hi, thanks for the report!
It works for me. Does this reproduce for you on a fresh clone? And does it still happen after you `cargo clean` your build?

L0g4n, 2022-10-21:
Damn, this is embarrassing. It worked after `cargo clean`. I assume the build artifacts of the `stable` branch messed it up.

closed 2022-10-21 by L0g4n

kaczmarczyck, 2022-10-21:
No worries, glad it works!