google/OpenSK #565: APDU parser has multiple errors

micolous, 2022-10-23:
1. APDU parser does not handle extended form with N<sub>c</sub> = 0 and N<sub>e</sub> > 0 ("Case 2E").  [It would error out with WRONG_LENGTH](https://github.com/google/OpenSK/blob/1b70583243a45392ca6638e891b3bbb965fd9be5/src/ctap/apdu.rs#L172-L176).

   Example request, where N<sub>c</sub> = 0 and N<sub>e</sub> = 0x1234:

   ```
   CLA INS P1 P2 0x00 0x12 0x34
   ```

   U2F GET_VERSION is impacted by this, other U2F commands and CTAP2 always have command payloads.

   The parser appears to only handle "Case 2S" (short form N<sub>c</sub> = 0 and N<sub>e</sub> > 0).

2. APDU parser accepts _malformed_ 3 byte L<sub>c</sub> + 3 byte L<sub>e</sub>.

   Per ISO 7816-4:2005 Section 5.1 (Command-response pairs), when N<sub>c</sub> > 0 and N<sub>e</sub> > 0 and using extended form ("Case 4E"), the correct representation is:

   ```
   CLA INS P1 P2 0x00 {Nc as uint16_be} {command bytes} {Ne as uint16_be}
   ```

    [The implementation here](https://github.com/google/OpenSK/blob/1b70583243a45392ca6638e891b3bbb965fd9be5/src/ctap/apdu.rs#L212-L224) appears to _also_ incorrectly accept _malformed_ requests of the form:

   ```
   CLA INS P1 P2 0x00 {Nc as uint16_be} {command bytes} 0x00 {Ne as uint16_be}
   ```

   The only reason L<sub>e</sub> should be prefixed with `0x00` is if the L<sub>c</sub> field is **absent** and L<sub>e</sub> is in extended form ("Case 2E").

**Note:** "Case 2E", "Case 2S" and "Case 4E" references to ISO 7816-3:2006 section 12.1.3 (Decoding conventions for command APDUs); S = short form, E = extended form.

Case | Command Data Field | Response Length Field | Total length
----- |  ------ | ---- | -----
Case 1 | None | None | 4 bytes
Case 2S | None | Present (1 byte) | 5 bytes
Case 2E | None | Present (3 bytes) | 7 bytes
Case 3S | Present (1 byte) | None | 5 + (command data length) bytes
Case 3E | Present (3 bytes) | None | 7 + (command data length) bytes
Case 4S | Present (1 byte) | Present (1 byte) | 6 + (command data length) bytes
Case 4E | Present (3 bytes) | Present (2 bytes) | 9 + (command data length) bytes

kaczmarczyck, 2022-10-24:
Hi, thanks for the writeup, this is useful!
Our NFC implementation is still work in progress, and these bugs will be great test cases when we continue working on it.

micolous, 2023-02-02:
By the way, after getting my hands on more FIDO2 keys, as far as I can tell pretty much only Yubikey implements the NFC side of things even *close* to correctly (there are bugs, just minor ones). Feitian, Hideez and Token2 keys don't implement extended L<sub>c</sub>/L<sub>e</sub> at all (as required by the CTAP spec), even though they are allegedly "certified"... so OpenSK is *already* doing better. 😄 

Relatedly, I wasn't able to find where OpenSK's ATR was, but that needs to flag its extended L<sub>c</sub>/L<sub>e</sub> in the "card capabilities" field (per ISO 7816-4 §8.1.1.2.7); again that's another case where only Yubikey seems to do this correctly.

kaczmarczyck, 2026-10-08:
The new APDU parser code needs to still be tested on hardware.