google/OpenSK #568: Pinned rust toolchain (nightly-2021-03-25) from libtock-rs causes linking errors on aarch64-apple-darwin

L0g4n, 2022-10-27:
## Expected Behavior

Compiling OpenSK should work fine on platforms like `aarch64-apple-darwin` (e.g. with `cargo build --features std`)

## Actual Behavior

Compiling with the pinned toolchain from [libtock-rs](https://github.com/tock/libtock-rs/blob/15e837e4495df2d7a39ae6d6c057fdec515cc430/rust-toolchain) causes linking errors on `aarch64-apple-darwin`. I assume that rust support for apple silicon chips was not ready yet at this point.

## Steps to Reproduce the Problem

1. Run `cargo build --features std` on `aarch64-apple-darwin` and observe the following C++ linking errors (watch out, very long).

>   Compiling libtock_core v0.1.0 (/Users/yk/Documents/Coding/opensk_fork/third_party/libtock-rs/core)
> error: linking with `cc` failed: exit status: 1
>   |
>   = note: "cc" "-arch" "arm64" "-L" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.0.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.1.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.10.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.11.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.12.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.13.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.14.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.15.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.2.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.3.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.4.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.5.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.6.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.7.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.8.rcgu.o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.build_script_build.z8vntg2w-cgu.9.rcgu.o" "-o" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e" "/Users/yk/rust-artifacts/debug/build/proc-macro2-f3cf9a69785a363e/build_script_build-f3cf9a69785a363e.3hj9l8a3kvh4kb7e.rcgu.o" "-Wl,-dead_strip" "-nodefaultlibs" "-L" "/Users/yk/rust-artifacts/debug/deps" "-L" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libstd-c74e6b3de522962e.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libpanic_unwind-c93ea4d419dae922.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libobject-d002150646c82cc7.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libaddr2line-08d0962d65625914.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libgimli-19ddd96688ea0319.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libstd_detect-d8646d3f69dbb0b4.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/librustc_demangle-53d2a1a1245eee83.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libhashbrown-66a22362d0e1778d.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/librustc_std_workspace_alloc-ad006e595a77ae48.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libunwind-0d9b6902b79f5775.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libcfg_if-b94ae0abe03da051.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/liblibc-e1883bf629bf5f4b.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/liballoc-43a6bd8ab40401f6.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/librustc_std_workspace_core-309925a076418223.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libcore-4ce201db55ae746f.rlib" "/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libcompiler_builtins-226f06aaeebfc929.rlib" "-lSystem" "-lresolv" "-lc" "-lm" "-liconv"
>   = note: ld: in /Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libpanic_unwind-c93ea4d419dae922.rlib(lib.rmeta), archive member 'lib.rmeta' with length 35728 is not mach-o or llvm bitcode file '/Users/yk/.rustup/toolchains/nightly-2021-05-25-aarch64-apple-darwin/lib/rustlib/aarch64-apple-darwin/lib/libpanic_unwind-c93ea4d419dae922.rlib' for architecture arm64
>           clang: error: linker command failed with exit code 1 (use -v to see invocation)

## Possible Solution

Updating the pinned nightly toolchain to a later version like `2021-06-25` and hence updating the specified submodule. With simple overwriting I found out that for example `nightly-2021-25` does not seem to work yet but the June version one month later indeed does (meaning the above compile command passed without any errors). Since the toolchain is propagated through the `libtock-rs` version we need to be careful not break any (runtime) functionality with later versions.

kaczmarczyck, 2022-10-28:
We use the nightly from `2021-06-25` on our [Hybrid PQC tag](https://github.com/google/OpenSK/releases/tag/hybrid-pqc), so upgrading should be possible. I didn't encounter any issues on this branch.

Does [this patch](https://github.com/google/OpenSK/blob/93357524d9012b4dba068a3690c7638cee643ec3/patches/libtock-rs/02-compiler-version.patch) fix your problem then?

I'll assign to @jmichelp to reproduce on Mac.

L0g4n, 2022-10-28:
Yes, I already managed to compile the develop branch on my arm64 Mac with the June nightly version.

micolous, 2023-02-07:
Same issue affects `x86_64-apple-darwin` (ie: on Intel macOS) while building TockOS `kernel`, running `./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer nordicdfu` at `stable` (1b70583), `develop` (684d37f), `hybrid-pqc` (9335752) on Ventura 13.2; which run with many different nightly toolchains.

Failures appear to flip between `libaddr2line-...rlib`, `libcfg_if-...rlib`, `librustc_std_workspace_core-...rlib`, and many other things.

I think `aarch64-apple-darwin` is a red herring - I think this might be triggered by a newer version of `clang`. Ventura 13.2 currently has `clang-14000.0.29.202` on `x86_64-apple-darwin-22.3.0`.

kaczmarczyck, 2023-05-06:
On `develop`, our toolchain is now from 2023. So this shuld be resolved?

L0g4n, 2023-05-06:
Yes, indeed.

closed 2023-05-06 by L0g4n