google/OpenSK #578: NRF52840 moduł RF E104-BT5040U

Firemanpl, 2023-01-06:
## Expected Behavior
This board on aliexpress
NRF52840 moduł RF E104-BT5040U

## Actual Behavior
Wsl doesn't recognize DFU mode. I was pressed resset and was plug into usb. I trying with VirtualBox with xubuntu also and result is the same. 
[image]

- Platform: DFU

Firemanpl, 2023-01-06:
[image]
[image]

Firemanpl, 2023-01-06:
Ok I solved you need to set permission to serial: 

`sudo chmod 666 /dev/ttyACM0`

Firemanpl, 2023-01-06:
HEH  not respond when plugged into usb after flashing. AWWW. Yes this is it what i expected. I guess that i bricked it. All leds not respond and doesn't see device.

Firemanpl, 2023-01-06:
[image]
[image]
[image]

Firemanpl, 2023-01-06:
[image]

Firemanpl, 2023-01-06:
[image]

syabyr, 2024-02-06:
This E104-BT5040U is a pca10059 clone.
you can use jlink or other link flash uf2 bootloader first:
``` bash
JLinkExe -If SWD -Speed 128 -Device NRF52840_XXAA
loadfile pca10059_bootloader-0.8.3_s140_6.1.1.hex
```

then build a nrf52840_dongle_dfu opensk firmware.
It works.

meltnoexit, 2024-08-23:
It's hard to carry it without a enclosure,but its original enclosure is useless because i can't press the buttons when it has the original enclosure on,so how to get a suitable enclosure？