google/OpenSK #584: MakeCredentialMissingParameterTest fails

bunnie, 2023-01-30:
## Expected Behavior
This test:

https://github.com/google/CTAP2-test-tool/blob/1afd50bd1b700dc86d6dffeca70a331771d24e45/src/tests/make_credential.cc#L593-L609

Should report a passing result, by denying the transaction.

## Actual Behavior

The transaction is allowed to proceed and return a credential response, as seen below:

```
Failed test: Tests if client PIN fails with missing parameters in MakeCredential. - Missing PIN parameters were not rejected when PIN is set.
A prompt was sent unexpectedly.
Expected error code `CTAP2_ERR_PIN_REQUIRED`, got `CTAP2_OK`.
The failing error code is `CTAP1_ERR_OTHER`.
```

## Steps to Reproduce the Problem

1. Run OpenSK `develop` branch against the CTAP2-test-tool
1. Search for the error message in the log

## Specifications

- Version: `develop` fork to betrusted-io
- Platform: Precursor

kaczmarczyck, 2023-02-07:
I read this note as a confirmation for what OpenSK is doing:
[image]

Regarding the transaction log, can you explain why rejection is the correct behavior for OpenSK?

bunnie, 2023-02-07:
Ah, the reason I think it's the correct behavior is because that's what the testbench was looking for.

Looking at your other responses, it seems more likely that the testbench is just out of date, and not a problem with your code, then.

bunnie, 2023-02-07:
I'll close this out, chalking this up to the test program being out of date...thanks!

closed 2023-02-07 by bunnie

kaczmarczyck, 2023-02-07:
Sorry for the confusion, and thanks for contributing!