google/OpenSK #586: GetAssertionOptionUvTrueTest test fails

bunnie, 2023-01-30:
## Expected Behavior

GetAssertionOptionUvTrueTest should pass

## Actual Behavior

```
Failed test: Tests is user verification set to true is accepted in GetAssertion. - The user verification option (true) was not accepted.
A prompt was expected, but not performed. Sometimes it is just not recognized if performed too fast.
The failing error code is `CTAP2_ERR_PIN_AUTH_INVALID`.
```

## Steps to Reproduce the Problem

1. Run the CTAP2-test-tool on the `develop` branch
1. Search for the error in the log

## Specifications

- Version: fork of `develop` branch
- Platform: Precursor

bunnie, 2023-01-30:
The test in question is this test: https://github.com/google/CTAP2-test-tool/blob/1afd50bd1b700dc86d6dffeca70a331771d24e45/src/tests/get_assertion.cc#L253-L280

This check can "never" pass, because the `permission` is consumed after a user presence is requested, since `self.client_pin.clear_token_flags()` is always called right after the permission is requested (see https://github.com/google/OpenSK/blob/684d37fa03ee65326006017e5d7314b76e87c44c/src/ctap/mod.rs#L815-L816).

The specific sequence that is observed in the failing test is:
 - SetPinToken is called to set a PIN
 - GetPinToken is called, which sets a default permission of `3`. This allows the next step, namely...
 - A test credential is created, requiring user presence. This consumes the permission previously granted by GetPinToken
 - Finally, GetAssertion is requested with `GetAssertionOptions { up: true, uv: true }` and pin_uv_auth_param: Some()...

This combination causes the following lines of code to return an error:

https://github.com/google/OpenSK/blob/684d37fa03ee65326006017e5d7314b76e87c44c/src/ctap/mod.rs#L1129-L1134

Because the permissions were previously cleared with `self.client_pin.clear_token_flags()` by the creation of the test credential.

I am not quite sure how to resolve this. Commenting out those checks causes the test bench to pass, but I do not feel confident enough about that as a fix to create a PR for it.

I can see a few scenarios:

- The checks cited above are inappropriate (fix: remove them)
- It is mandatory to do a GetPinToken immediately prior to getting an assertion, so that permissions can be established. I don't see anything about this in the spec, however, so I think this isn't the likely explanation.
- Something is wrong in the consumption of permissions after the user presence test. Perhaps a less expansive set of flags should be gated off after user presence?

Looking back at the `stable` branch for OpenSK, it seems that perhaps the reason this test passed before is because the
 permissions are not cleared at all by the user presence test: there is no equivalent of `self.client_pin.clear_token_flags();`
 after the user presence test:

https://github.com/google/OpenSK/blob/1b70583243a45392ca6638e891b3bbb965fd9be5/src/ctap/mod.rs#L508-L510

You can see that the code simply moves on, and eventually the permissions will expire due to the periodic timer that will clear the permissions.

I strongly suspect the answer is the last one, that too many permissions are cleared after the user presence test, but I am not familiar enough with the permissions state machine to make that call, hence I am logging this issue to request your advice.

thanks!

kaczmarczyck, 2023-02-07:
Thanks for the thorough report. The CTAP2 Test Tool is not yet up to date with FIDO 2.1. I think OpenSK is correctly calling `clearPinUvAuthTokenPermissionsExceptLbw` in both MakeCredential and GetAssertion. This matches the specification.

The tests were written when we didn't yet need a fresh auth token after each call to Make and Get. There is already an issue for this on the [Test Tool](https://github.com/google/CTAP2-test-tool/issues/111). So I'll close this issue here. Feel free to ignore the failing test.

closed 2023-02-07 by kaczmarczyck

bunnie, 2023-02-07:
Great, thanks!