google/OpenSK #592: "configuring device" step needs a confirmation prompt to replug device on nrf52840_dongle_dfu

micolous, 2023-02-11:
## Expected Behaviour

Running `./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer dfu` successfully configure the device.

It looks like the nRF52840 dongle needs to be power cycled after DFU (pressing the sideways reset button is not enough) to boot into the flashed firmware successfully.  There should be a confirmation prompt like at the "flashing" step to get the user to replug the device.

## Actual Behaviour

```
$ python3.10 ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer nordicdfu
info: Updating rust toolchain to nightly-2021-03-25
info: syncing channel updates for 'nightly-2021-03-25-x86_64-unknown-linux-gnu'
info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
info: Rust toolchain up-to-date
Done: The submodules and patches are in sync.
info: Building Tock OS for board nrf52840_dongle_dfu
   Compiling kernel v0.1.0 (/home/michael/opensk/third_party/tock/kernel)
   Compiling cortexm v0.1.0 (/home/michael/opensk/third_party/tock/arch/cortex-m)
   Compiling nrf5x v0.1.0 (/home/michael/opensk/third_party/tock/chips/nrf5x)
   Compiling capsules v0.1.0 (/home/michael/opensk/third_party/tock/capsules)
   Compiling cortexm4 v0.1.0 (/home/michael/opensk/third_party/tock/arch/cortex-m4)
   Compiling nrf52 v0.1.0 (/home/michael/opensk/third_party/tock/chips/nrf52)
   Compiling nrf52840 v0.1.0 (/home/michael/opensk/third_party/tock/chips/nrf52840)
   Compiling components v0.1.0 (/home/michael/opensk/third_party/tock/boards/components)
   Compiling nrf52_components v0.1.0 (/home/michael/opensk/third_party/tock/boards/nordic/nrf52_components)
   Compiling nrf52840_dongle_dfu v0.1.0 (/home/michael/opensk/third_party/tock/boards/nordic/nrf52840_dongle_dfu)
    Finished release [optimized + debuginfo] target(s) in 14.33s
info: Building OpenSK application
Testing invariants in customization.rs...
   Compiling libtock_core v0.1.0 (/home/michael/opensk/third_party/libtock-rs/core)
   Compiling libtock_drivers v0.1.0 (/home/michael/opensk/third_party/libtock-drivers)
   Compiling rng256 v0.1.0 (/home/michael/opensk/libraries/rng256)
   Compiling lang_items v0.1.0 (/home/michael/opensk/third_party/lang-items)
   Compiling crypto v0.1.0 (/home/michael/opensk/libraries/crypto)
   Compiling ctap2 v1.0.0 (/home/michael/opensk)
    Finished test [unoptimized + debuginfo] target(s) in 5.32s
     Running unittests (target/debug/deps/ctap2-ef011d56510e2013)
   Compiling libtock_core v0.1.0 (/home/michael/opensk/third_party/libtock-rs/core)
   Compiling libtock_drivers v0.1.0 (/home/michael/opensk/third_party/libtock-drivers)
   Compiling rng256 v0.1.0 (/home/michael/opensk/libraries/rng256)
   Compiling lang_items v0.1.0 (/home/michael/opensk/third_party/lang-items)
   Compiling crypto v0.1.0 (/home/michael/opensk/libraries/crypto)
   Compiling ctap2 v1.0.0 (/home/michael/opensk)
    Finished release [optimized] target(s) in 9.63s
info: Generating Tock TAB file for application/example ctap2
info: Generating all-merged HEX file: target/nrf52840_dongle_dfu_merged.hex
info: Creating DFU package
info: Please insert the dongle and switch it to DFU mode by keeping the button pressed while inserting...
info: Press [ENTER] when ready.

info: Flashing device using DFU...
  [####################################]  100%
Device programmed.
info: Configuring device.
fatal: No device to configure found.
```

If you replug the dongle quickly enough after `info: Configuring device`, you instead get the correct outcome:

```
info: Generating Tock TAB file for application/example ctap2
info: Generating all-merged HEX file: target/nrf52840_dongle_dfu_merged.hex
info: Creating DFU package
info: Please insert the dongle and switch it to DFU mode by keeping the button pressed while inserting...
info: Press [ENTER] when ready.

info: Flashing device using DFU...
  [####################################]  100%
Device programmed.
info: Configuring device.
-- REPLUGGED DEVICE HERE --
info: Programming OpenSK device AAGUID 3b844af5-18ba-412f-9a65-fc50dba80bac (CtapHidDevice('/dev/hidraw8')).

info: Certificate: Missing

info: Private Key: Missing

100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 24.54it/s]
info: Your device is not yet configured, and lacks some functionality. If you run into issues, this command might help:

./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key

Please read the Certificate considerations in docs/customization.md to understand the privacy trade-off.
```

## Specifications

- Version: a222986
- Platform: nrf52840_dongle_dfu

kaczmarczyck, 2023-02-13:
I can't reproduce. This is how the command ends for me without replugging:

```
info: Generating Tock TAB file for application/example ctap2
info: Generating all-merged HEX file: target/nrf52840_dongle_dfu_merged.hex
info: Creating DFU package
info: Please insert the dongle and switch it to DFU mode by keeping the button pressed while inserting...
info: Press [ENTER] when ready.

info: Flashing device using DFU...
  [####################################]  100%          
Device programmed.
info: Configuring device.
info: Programming OpenSK device AAGUID fab4ba9b-328b-4c62-bb39-52372a9896c3 (CtapHidDevice('/dev/hidraw1')).                                
info: Certificate: Missing                                                                                                                  
info: Private Key: Missing                                                                                                                  
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 26.65it/s]
info: Your device is not yet configured, and lacks some functionality. If you run into issues, this command might help:

./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key

Please read the Certificate considerations in docs/customization.md to understand the privacy trade-off.
```

I use a ThinkPad running Linux. Since we have a retry loop that tries over a span of 10 seconds, it shouldn't be a speed problem, but the device doesn't ever boot before you replug. I.e. if you call configure manually as instructed nothing happens even after a minute?

micolous, 2023-03-01:
In that setup, I was using a 2014 MacBook Pro running Linux natively, but the dongle was connected to a powered USB 3.0 hub (built-in to an external monitor). I'll have to double check when I'm back home what was happening on the USB side in `dmesg`, but from memory the device didn't come back until I actually power cycled it.

micolous, 2023-03-02:
I've run another flash at d8512b4417032e58182e1ad02863216893657236, and still see the issue when connected to the monitor's powered USB hub.  I only plugged in the dongle when prompted to, and in `dmesg` I get:

```
[  311.014080] usb 1-1.4: new full-speed USB device number 19 using xhci_hcd
[  311.117840] usb 1-1.4: New USB device found, idVendor=1915, idProduct=521f, bcdDevice= 1.00
[  311.117856] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  311.117861] usb 1-1.4: Product: Open DFU Bootloader
[  311.117866] usb 1-1.4: Manufacturer: Nordic Semiconductor
[  311.117870] usb 1-1.4: SerialNumber: (redacted)
[  311.139110] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[  311.139154] usbcore: registered new interface driver cdc_acm
[  311.139157] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  332.808403] usb 1-1.4: USB disconnect, device number 19
```

The disconnect occurs when DFU finishes. I waited 5 minutes, and the device did not came back until I manually unplugged and re-plugged the dongle:

```
[  659.431452] usb 1-1.4: new full-speed USB device number 20 using xhci_hcd
[  659.534975] usb 1-1.4: New USB device found, idVendor=1915, idProduct=521f, bcdDevice= 0.01
[  659.534990] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  659.534996] usb 1-1.4: Product: OpenSK
[  659.535001] usb 1-1.4: Manufacturer: Nordic Semiconductor ASA
[  659.535005] usb 1-1.4: SerialNumber: v1.0
[  659.544532] hid-generic 0003:1915:521F.000E: hiddev3,hidraw8: USB HID v1.10 Device [Nordic Semiconductor ASA OpenSK] on usb-0000:00:14.0-1.4/input0
```

I also tried connecting the dongle to one of the USB ports on the MacBook itself to reflash it, and see the same behaviour.

The only USB controller in this MacBook is:

```
00:14.0 USB controller [0c03]: Intel Corporation 8 Series USB xHCI HC [8086:9c31] (rev 04)
```

The hub is in-built to a 1080p touchscreen display with 4 physical ports, which I have connected via a DisplayPort and a USB cable:

```
USB 3.0:
Bus 002 Device 006: ID 0424:5734 Microchip Technology, Inc. (formerly SMSC) USB5734   [root hub, 4 ports]

USB 2.0:
Bus 001 Device 007: ID 0424:2734 Microchip Technology, Inc. (formerly SMSC) USB2734   [root hub, 5 ports]
Bus 001 Device 009: ID 0424:2412 Microchip Technology, Inc. (formerly SMSC) Hub  [touchscreen and hub controller hub, 2 ports]
Bus 001 Device 010: ID 0424:274c Microchip Technology, Inc. (formerly SMSC) Hub Controller
Bus 001 Device 011: ID 1fd2:8105 Melfas LGDisplay Incell Touch  [HID touchscreen]
```

I don't have this machine connected to a Thunderbolt dock, and I don't have a Thunderbolt 2 dock to try an external USB controller.

The only USB weirdness on this machine is it fails to initially enumerate the internal keyboard and trackpad when booting or reloading the `xhci-pci` kernel module:

```
[   12.527655] usb 2-3: device not accepting address 3, error -62
[   17.951756] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   23.327721] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[   23.535684] usb 2-3: device not accepting address 4, error -62
[   23.543795] usb usb2-port3: attempt power cycle
[   23.671688] usb 1-5: new full-speed USB device number 4 using xhci_hcd
[   23.825887] usb 1-5: New USB device found, idVendor=05ac, idProduct=0259, bcdDevice= 2.24
[   23.825902] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   23.825908] usb 1-5: Product: Apple Internal Keyboard / Trackpad
[   23.825912] usb 1-5: Manufacturer: Apple Inc.
```

This also happens when unplugging and re-plugging the USB hub, it takes about 30 seconds before it is fully usable.

However, non-hub devices connected to the laptop directly or to an external hub are not impacted, they enumerate in about 1 second, and there are no timeout/reset warnings. Similarly, manually re-plugging the dongle after DFU within that 10 second window works just fine.

Unfortunately, because of #568 I can't run it on the other machine I'd try to use it on (that issue also affects x86_64), and I couldn't get USB pass-through working with virtualisation... macOS really doesn't want to release the dongle for use by my VM. 😞 

I think that debugging it further with that hardware will need a logic analyser; my suspicion is that the nRF52840 depends on a USB power cycle or reset that is not happening.  The work-around for me is to manually re-plugging the device as soon as DFU is done; which is why I thought putting a confirmation prompt (or explicitly probing for the USB device coming back) would make that process more reliable.

USB timing would be even more temperamental if I were successful at getting this running in a VM, as that sometimes requires clicking confirmation windows or menus to re-attach a "new" USB device to the VM.

kaczmarczyck, 2023-03-03:
The configuration step is always somewat manual anyway. It reads your configuration, and then gives you a recommendation:

- If your device is configured, nothing to do.
- If not, it prints the command you'd have to run if you wanted to configure a certificate.

It doesn't automatically do that because of the privacy implications. In case of failure to enumerate, we could also decide to print more information before exit? The command is just

```
./tools/configure.py [--vendor-hid]
```

to print information about the configuration status, and

```
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
```

to set the certificate. Would printing instructions in case of error be a solution for you?

micolous, 2023-03-09:
> Would printing instructions in case of error be a solution for you?

That would work, thanks!

PR #604 (2023-03-11): Fixes configuration failure modes in deploy.py
Fixes #592

This PR does 3 small fixes:

- Implement the suggestion from [this comment](https://github.com/google/OpenSK/issues/592#issuecomment-1453425199)
- Don't try to configure examples
- Correctly hint the configure command needed with `vendor_hid`
closed 2023-03-11 by kaczmarczyck