google/OpenSK #619: error: failed to get `arrayref` as a dependency of package `ctap2 v1.0.0 (/home/USER/OpenSK)`

trustkeysolutions, 2023-04-26:
I am getting the some errors when deploying OpenSK to nRF52840 dongle. I've added the information below. Any help would be appreciated it resolving this error so that I can get OpenSK running on the dongle. Thanks.

## Expected Behavior
Expect to have deployed OpenSK to nRF52840 dongle after running the deployment

## Actual Behavior
Getting the following error:
error: failed to get `arrayref` as a dependency of package `ctap2 v1.0.0 (/home/USER/OpenSK)`

Caused by:
  failed to load source for dependency `arrayref`

Caused by:
  Unable to update registry `https://github.com/rust-lang/crates.io-index`

Caused by:
  failed to fetch `https://github.com/rust-lang/crates.io-index`

Caused by:
  error reading from the zlib stream; class=Zlib (5)
fatal: Failed to execute cargo: Command '['cargo', 'build', '--release', '--target=thumbv7em-none-eabi', '--features=with_ctap1']' returned non-zero exit status 101.

## Steps to Reproduce the Problem

1. Install Ubuntu on Windows 11 (wsl  --install)
2. Start Ubuntu 
3. Update Ubuntu by running the following commands:
  $sudo apt update
  $sudo apt upgrade
4. Install the required dependencies:
  $sudo apt install -y git python3 curl
5. Install Rust:
  $curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
6. Clone the repository:
  $git clone https://github.com/google/OpenSK.git
7. Install the required Rust targets and tools from /home/USER/OpenSK:
  $./setup.sh
8. Install additional dependencies
  $sudo apt install python3-pip
  $pip3 install --user colorama
  $pip3 install --user tockloader==1.5.0
  $pip3 install --user nrfutil
8. Deploy
  $./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu

## Specifications

- Version: 22.04
- Platform: Ubuntu 22.04.2 LTS

kaczmarczyck, 2023-04-26:
Hi! This looks like a broader Rust setup problem to me. Quick question in case you haven't tried that: Can you run any example problem using `array_ref` (i.e., not OpenSK)? Or compile any Rust that imports crates?

trustkeysolutions, 2023-04-26:
Thanks for your response, @kaczmarczyck. I've created a simple example:

-Cargo.toml
[dependencies]
arrayref = "0.3.6"

-src/main.rs:
extern crate arrayref;

fn main() {
    let data: [i32; 8] = [0, 1, 2, 3, 4, 5, 6, 7];
    let (left_half, right_half) = arrayref::array_refs!(&data, 4, 4);
    println!("Left half: {:?}", left_half);
    println!("Right half: {:?}", right_half);
}

-output from cargo run:
    Updating crates.io index
  Downloaded arrayref v0.3.7
  Downloaded 1 crate (9.6 KB) in 0.40s
   Compiling arrayref v0.3.7
   Compiling arrayref_example v0.1.0 (/home/USER/arrayref_example)
    Finished dev [unoptimized + debuginfo] target(s) in 1m 19s
     Running `/home/USER/arrayref_example/target/debug/arrayref_example`
Left half: [0, 1, 2, 3]
Right half: [4, 5, 6, 7]

trustkeysolutions, 2023-04-26:
It was able to build after a clean build using tmux terminal on ubuntu. I am not sure why it worked in a different env. I'll close this ticket. Thanks for your time.

closed 2023-04-26 by trustkeysolutions

kaczmarczyck, 2023-04-26:
I am happy you found your problem!

kaczmarczyck, 2023-06-13:
I got a report from another user that they ran into the same problem. The solution seemed to be noon OpenSK specific, but the only general issue I found is this:
https://github.com/rust-lang/cargo/issues/10303

It suggests running cargo with [`CARGO_NET_GIT_FETCH_WITH_CLI`](https://doc.rust-lang.org/cargo/reference/config.html#netgit-fetch-with-cli)

@trustkeysolutions Can I ask you to add more details about how you fixed the problem back then?

phd2018eth, 2023-06-14:
An identical problem as @trustkeysolutions reported except I am not running wsl but in both Ubuntu 22.04  VM nad native modes. As @kaczmarczyck suggested, it was fixed by setting CARGO_NET_GIT_FETCH_WITH_CLI to true.