google/OpenSK #621: WebUSB compatible?

trustkeysolutions, 2023-04-28:
## Expected Behavior
Be able to connect via WebUSB to nRF52840 dongle running OpenSK 

## Actual Behavior
Getting the following error: Error connecting to FIDO device: DOMException: Failed to execute 'open' on 'USBDevice': Access denied.

## Steps to Reproduce the Problem

1. Create a react app, install webusb and run the following App

// ./src/App.tsx

import React, { useState } from "react";
import "./App.css";

const App: React.FC = () => {
  const [connected, setConnected] = useState<boolean>(false);
  const [device, setDevice] = useState<USBDevice | null>(null);
  const [message, setMessage] = useState<string>("");

  const connectToFidoDevice = async () => {
    if (!connected) {
      try {
        const filters = [
          {
            vendorId: 6421, // OpenSK
            productId: 21023,
          },
        ];
        const requestedDevice = await navigator.usb.requestDevice({ filters });
        await requestedDevice.open();

        setDevice(requestedDevice);
        setConnected(true);
        setMessage("Successfully connected and opened FIDO Device");
      } catch (error) {
        setMessage("Error connecting to FIDO Device: " + error.message);
      }
    } else {
      if (device) {
        await device.close();
        setConnected(false);
        setMessage("Disconnected from FIDO Device");
      }
    }
  };

  return (
    <div className="App">
      <h1>OpenSK Connection</h1>

      <button onClick={connectToFidoDevice} style={{ marginBottom: "10px" }}>
        {connected
          ? "Disconnect from FIDO Device"
          : "Connect to FIDO Device"}
      </button>

      <div className="message">{message}</div>
    </div>
  );
};

export default App;

## Specifications

- Version: OpenSK stable
- Platform: Version 112.0.5615.138 (Official Build) (64-bit)

jmichelp, 2023-04-28:
The FIDO HID page is locked down by operating systems for security reasons. The fact that you can't connect to a security key using WebUSB is working as intended.

closed 2023-04-28 by jmichelp

kaczmarczyck, 2023-05-02:
What do you intend to achieve through WebUSB? We have a `--vendor-hid` flag on develop for this purpose, and it allows you to run non-FIDO commands through WebUSB.

trustkeysolutions, 2023-05-02:
My goal for utilizing WebUSB is to facilitate seamless communication between a web application and the user's FIDO device. By leveraging WebUSB, I was trying to see if we can improve user experience with a more straightforward device setup and management process via the browser instead of a desktop app.

Thanks for the info on --vendor-hid flag. This flag indeed allows running non-FIDO commands through WebUSB, which could be useful for our application. However, my current focus is primarily on implementing FIDO-related functionality, but sounds like this is a dead-end. 

That said, we will keep the --vendor-hid flag in mind for future development stages, especially when expanding our application to support non-FIDO features or when working with vendor-specific commands. Thanks, again.

kaczmarczyck, 2023-05-02:
Warning to all future users that come here: The following advice is dangerous if done wrong, please read to the end.

If you don't mind patching the OpenSK logic (and use the develop branch that is currently worked on), it's not hard to allow FIDO commands over vendor HID. The switch that is responsible is here:
https://github.com/google/OpenSK/blob/645c1ba3a7275dce7ea6b04a344df0a7559690c9/libraries/opensk/src/ctap/mod.rs#L603

Remove the match over `channel`, always call `process_fido_command` and OpenSK allows FIDO calls over WebUSB.

There is a reason why we don't allow this, though: Anyone with WebUSB access can then freely send CTAP requests to your device. That includes silent authentication, which means any website could impersonate the user cross origin. I strongly suggest you don't do that, but instead have an authenticated vendor command. FIDO's security model relies on the browser being trustworthy, and therefore opening a hole to circumvent the browser is a bad idea.

Doing some management over the vendor HID channel should be fine though. For example, we send signed upgrades through that channel. The signature guarantees that we can't be impersonated.