google/OpenSK #648: nordic dfu in develop fails to configure

PhearZero, 2023-09-15:
## Expected Behavior

Expect configuration to be successful
```bash
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
```
## Actual Behavior
```bash
info: Flashing device using DFU...
  [####################################]  100%          
Device programmed.
info: Configuring device.
fatal: No device to configure found.
```

## Steps to Reproduce the Problem

1. checkout `develop`
2. ./setup.sh
3. ./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu

## Specifications

- Platform: Nordic NRF5284 Dongle

ia0, 2023-09-15:
Thanks for the report! This looks similar to #592. Did you try running the following after `deploy.py`?

```
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
```

Ideally it should be printed in the error message, but according to #592 it looks like there might be some race condition that doesn't print the instructions.

PhearZero, 2023-09-15:
You are welcome! It seems like they are similar and can confirm the race condition. I have gotten the `configure.py` message a few times but not every time. After the DFU flash, I get the following error while running configure:

```bash
./tools/configure.py --certificate=crypto_data/opensk_cert.pem  --private-key=crypto_data/opensk.key
info: Private key is valid.
info: Certificate is valid.
Traceback (most recent call last):
  File "/home/zero/IdeaProjects/OpenSK/./tools/configure.py", line 221, in <module>
    main(parser.parse_args())
  File "/home/zero/IdeaProjects/OpenSK/./tools/configure.py", line 135, in main
    devices = get_opensk_devices(args.batch)
  File "/home/zero/IdeaProjects/OpenSK/./tools/configure.py", line 67, in get_opensk_devices
    return [ctap2.Ctap2(dev)]
  File "/home/zero/.local/lib/python3.10/site-packages/fido2/ctap2/base.py", line 208, in __init__
    self._info = self.get_info()
  File "/home/zero/.local/lib/python3.10/site-packages/fido2/ctap2/base.py", line 262, in get_info
    return Info.from_dict(self.send_cbor(Ctap2.CMD.GET_INFO))
  File "/home/zero/.local/lib/python3.10/site-packages/fido2/ctap2/base.py", line 241, in send_cbor
    raise CtapError(status)
fido2.ctap.CtapError: CTAP error: 0xF2 - UNKNOWN
```

ia0, 2023-09-15:
This looks like an internal error in the GetInfo command. I don't see much reasons except a storage error. Could you try to erase the storage before flashing OpenSK? The stable and develop branches probably don't have compatible storage.

```
./deploy.py --board=nrf52840_dongle_dfu --erase_storage --programmer=nordicdfu
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
./tools/configure.py --certificate=crypto_data/opensk_cert.pem  --private-key=crypto_data/opensk.key
```

PhearZero, 2023-09-15:
That was the problem, thank you so much @ia0!

closed 2023-09-15 by PhearZero