google/OpenSK #667: Suggested environment to be able to build/compile

AussieMakerGeek, 2023-12-14:
## Expected Behavior
To be able to Build and compile the code

## Actual Behavior
Various errors under multiple linux based operating systems - I have tried Ubuntu 22.04, 18.04, PopOS etc but all seem to have issues, I even tried building under the latest Raspberry Pi OS. None are able to even get past Setup.sh.

18.04 - Fails at FIDO2 not able to be >1.0.0
22.04 - Unable to find a tockloader module

The instructions on the certificate are unclear and the command does not work

How is anyone even building this? Please suggest an environment where I should be able to build it.

Why can this not be provided as a binary that you can just flash since we're using generic/known hardware?  What am I missing here?

kaczmarczyck, 2023-12-14:
Hi!

What branch are you trying to build? The `2.1` branch and `develop` branch build for me locally on a Debian based OS. Our unit tests are running on GitHub under `ubuntu-latest` and `macos-latest`. At the moment, this means that it builds on Ubuntu 22.04.

Your errors sounds like you get an error on the `pip install` step for tockloader? Can you provide a more detailed log for what you tried to do?

There is no blocker for providing a binary. One way to do that could be a workflow that automatically adds one to a release. One dependency are reproducible builds, if we want people to trust our binaries. I added a comment to #182, but it's not a requirements, as long as we are transparent about it.

AussieMakerGeek, 2023-12-15:
Ok, So i tried a fresh install of Ubuntu 22.04 with branch 2.1 and was able to get as far as the deploy this time. The setup.sh seemed to run ok.

./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
fatal: Coun't load python3 module nordicsemi.lister, Try to run: pip3 install nordicsemi.lister

python --version
Python 3.10.12

The certificate configure.py gives:
info: Private key is valid.
info: Certificate is valid,
fatal: No devices found.

But it's not clear if I need to image it first, and then install the certificates or if the certificates are needed for the build.  The latter makes sense but this is not in the order of the howto.

kaczmarczyck, 2023-12-15:
If you want to flash a dongle, then you need `nrfutil`. Nordic stopped offering it on pip, so you need to manually install and use it. See:
- https://www.nordicsemi.com/Products/Development-tools/nrf-util
- https://github.com/NordicSemiconductor/pc-nrfutil/issues/366

We also discuss this on our issue #588. We will need to at least document ways to fix this problem. Things that might work (I haven't tried):

- Install python3.10, change the setup to ues that in venv and add `nrfutil` to requirements.txt before setup.
- Call the respective functions of the manually downloaded new Nordic tool.
- The suggestions in #588.

kaczmarczyck, 2023-12-15:
If you patch #669 (or wait for us to merge), you only need `nrfutil` version 6. You can either download the [nrfutil binary](https://github.com/NordicSemiconductor/pc-nrfutil/releases) or get it through pip if you have Python3.10 or older.

AussieMakerGeek, 2023-12-17:
Thanks for that - I did read that instruction but I made the assumption that 3.10.x would still be ok but from what you're saying it's strictly 3.10.

I was able to export the hex (programmer=none) and attempted to use the latest NRFutil binary (https://www.nordicsemi.com/Products/Development-tools/nrf-util) on both Linux and Windows but both are still erroring.  I'll have a crack with the version you linked.

AussieMakerGeek, 2023-12-17:
Same error with the linked nrfutil:
.\nrfutil.exe dfu serial -pkg .\nrf52840_dongle_dfu_merged.hex
Traceback (most recent call last):
  File "nordicsemi\__main__.py", line 1555, in <module>
  File "click\core.py", line 1137, in __call__
  File "click\core.py", line 1062, in main
  File "click\core.py", line 1668, in invoke
  File "click\core.py", line 1668, in invoke
  File "click\core.py", line 1404, in invoke
  File "click\core.py", line 763, in invoke
  File "nordicsemi\__main__.py", line 1073, in serial
  File "nordicsemi\__main__.py", line 968, in do_serial
  File "nordicsemi\lister\device_lister.py", line 67, in get_device
AttributeError: 'NoneType' object has no attribute 'lower'
[25332] Failed to execute script '__main__' due to unhandled exception!

I might have to flash it with openOCD

kaczmarczyck, 2023-12-18:
Agreed, our documentation isn't updated to the latest nrfutil yet. Do you mind if I rename this issue to "Flash dongle with latest nrfutil" so others can find and benefit from your efforts?

Other than that, feel free to close this issue when your problem is solved. Also a PR that updates our documentation is always appreciated, otherwise I will look into that when I find the time to replicate it on Linux.

AussieMakerGeek, 2023-12-19:
For sure - I'm already intending to at least document the full process that I used for the benefit of others.  I also utilized the E104-BT5040U dongle which is a little more complicated due to needing the generic DFU bootloader to be flashed - Which i used OpenOCD/Raspberry Pi for.  This dongle is 1:1 for the official NRF dongle as far as pin mappings. One thing that concerned me in the process was (apparently) the internal voltage regulator's register is reset with the DFU reflash but on this particular dongle it would appear that it's not utilized (or perhaps the DFU flash doesn't reset it but that's contrary to what I read). I assume it has an external voltage regulator but I didn't check it under my scope.

I also wired in a TTP223 capacitive touch sensor which JUST fits inside the case and makes it a perfect setup at a pretty good price point.

closed 2023-12-19 by AussieMakerGeek

kaczmarczyck, 2023-12-19:
Thanks! Maybe #607 is useful to you, if you use the E104-BT5040U.

AussieMakerGeek, 2023-12-19:
It did help, as did #545  and a couple of other external sources.  It's not that it's specifically hard, just the fact that the information is a little spread out and tends to have a higher level of assumed knowledge which can make the hurdle seem a little higher to jump.

Being that these dongles are relatively cheap and will work with no mods to the code, it makes sense to at least have a documented procedure to get them working, even if not officially supported by the project.