google/OpenSK #672: deploy.py succeeds programming but testing board not recognised by testing webpages.

Phizicks, 2023-12-28:
## Expected Behavior
Plugging in USB, expect flashing LED or at least testing webpage to detect dongle when button pressed

## Actual Behavior
after plugging USB, no flashing LED and device not recognised as device that can be used. pressing button regardless, nothing happens.

I am possibly expecting it's something to do with tockloader needing installing possibly. but when I tried `tockloader install`
it just hangs at "installing app on the board"
```
[INFO   ] No TABs passed to tockloader.
[STATUS ] Searching for TABs in subdirectories.
[INFO   ] Using: ['./target/tab/ctap2.tab']
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] No serial port with device name "tock" found.
[INFO   ] Found 2 serial ports.
[INFO   ] Using "/dev/ttyS0 - ttyS0".
[INFO   ] Using serial channel to communicate with the board.
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] No serial port with device name "tock" found.
[INFO   ] Found 2 serial ports.
Multiple serial port options found. Which would you like to use?
[0]	/dev/ttyS0 - ttyS0
[1]	/dev/ttyACM0 - Open DFU Bootloader

Which option? [0] 1
[INFO   ] Using "/dev/ttyACM0 - Open DFU Bootloader".
[STATUS ] Installing app on the board...
...

## Steps to Reproduce the Problem

https://github.com/google/OpenSK/blob/2.0/docs/boards/nrf52840_dongle.md
1. checkout repo and followed https://github.com/google/OpenSK/blob/2.1/docs/install.md
2. followed https://github.com/google/OpenSK/blob/2.0/docs/boards/nrf52840_dongle.md
```
 ./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
```
Note: can't run ./deploy.py --board=nrf52840_dongle --programmer=jlink due to no JLinkExe error
3. installed using
```
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
```
results
```
...
info: Flashing device using DFU...
  [####################################]  100%          
Device programmed.
info: Configuring device.
info: Programming OpenSK device AAGUID db510839-25a6-4424-a33b-9c800383b5c1 (CtapHidDevice('/dev/hidraw5')).                                                                                                                                                                             
info: Certificate: Present                                                                                                                                                                                                                                                               
info: Private Key: Present                                                                                                                                                                                                                                                               
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 22.71it/s]
info: You're all set!
...

## Specifications

- Version: 23.10
- Platform: Linux Ubuntu

kaczmarczyck, 2023-12-28:
I'm not sure I understand your steps to reproduce:

- One of the prompts seems to not work, and the other does?
- The linked documentation is sometimes referring to the 2.0 branch, and sometimes to 2.1. See for example steps 1 and 2. Maybe you can try a clean install for branch 2.1?

After the last `You're all set!`, your dongle should work, even if you plug it into an unrelated computer.
I don't understand at what point you used tockloader for what purpose. If you take the final device, and plug it into your Ubuntu machine, does it enumerate with `lsusb`?

Phizicks, 2023-12-28:
yeah sorry, I tried to follow the steps but then started to jump to other doc pages since I thought I missed steps that were required.
also noticed I had checked out the `develop` branch so switched to 2.1 and started clean using.

* switched to 2.1 and reset all
```
./reset.sh
./setup.sh

sudo cp rules.d/55-opensk.rules /etc/udev/rules.d/
sudo udevadm control --reload

```
* replugged in USB dongle, then resetup the certs
* LEDs were flashing, pressed the button to reset

```
./tools/configure.py \
    --certificate=crypto_data/opensk_cert.pem \
    --private-key=crypto_data/opensk.key
...
info: Private key is valid.
info: Certificate is valid.
info: Programming OpenSK device AAGUID db510839-25a6-4424-a33b-9c800383b5c1 (CtapHidDevice('/dev/hidraw1')).                                                                                                                                                                  
info: Please touch the device to confirm...                                                                                                                                                                                                                                   
info: Certificate: Present                                                                                                                                                                                                                                                    
info: Private Key: Present                                                                                                                                                                                                                                                    
100%|█...█| 1/1 [00:26<00:00, 26.30s/it]
```

then switched to rf52850 dongle page/steps as instructed at https://github.com/google/OpenSK/blob/2.1/docs/boards/nrf52840_dongle.md

now ran 
```
./deploy.py --board=nrf52840_dongle_dfu --opensk --programmer=nordicdfu
fatal: Couldn't load python3 module nordicsemi.lister. Try to run: pip3 install nordicsemi.lister
```

strange enough, I got a new error that the pip library `nordicsemi` isn't installed and I can't seem to install it though I do have `nrfutil` package and not sure why it's now asking this when it worked before.
```
pip3 install nordicsemi
ERROR: Could not find a version that satisfies the requirement nordicsemi (from versions: none)
ERROR: No matching distribution found for nordicsemi
```

I'm using conda environment of python 3.10

kaczmarczyck, 2023-12-29:
Ah yes, that is because of #669 on develop. I merged it because someone else had trouble with the package. The error message `Try to run: pip3 install nordicsemi.lister` is misguided, the package to install is `nrfutil`. Note that `setup.sh` installs with virtualenv, so maybe there is some confusion there? If `develop` works for you, then that's also totally okay! It is very close to 2.1.

When you call `./tools/configure.py`, it seems like you have a working OpenSK already, and there is nothing to do. What is it you are trying to accomplish? If you open [webauthn.io](https://webauthn.io/), can you use your OpenSK?

Phizicks, 2023-12-29:
> If you open [webauthn.io](https://webauthn.io/), can you use your OpenSK?
No, that is my main point. the leds don't trigger, and pressing the button on registration doesn't do anything. as if it's dead. this was the reason the warning of tockos I thought was a missing piece as maybe it wasn't booting properly.

when i restarted from scratch and ran the reset and so on, it did prompt me to press the button and I did have the 2 leds alternatingly flashed so it seems to be accessible at least by code.

I am using FF 121.0 on ubuntu 23.10 desktop if that helps.

Phizicks, 2023-12-29:
configure output
```
(py310) eitadmin@cipher:~/workspace/code/OpenSK$ ./tools/configure.py
info: Programming OpenSK device AAGUID db510839-25a6-4424-a33b-9c800383b5c1 (CtapHidDevice('/dev/hidraw1')).                      
info: Certificate: Present                                                                                                        
info: Private Key: Present                                                                                                        
100%|███████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00, 25.45it/s]
```

kaczmarczyck, 2023-12-29:
If it blinks and reacts to buttons, it is very unlikely that there is a problem with your OpenSK.
I don't see how webauthn.io shouldn't work, it interacts fine with Firefox for me. To make sure your authenticator works, you can run [this example from libfido2](https://github.com/Yubico/python-fido2/blob/main/examples/credential.py). If that runs without problems, it is a OS / platform problem.

Phizicks, 2023-12-30:
so did I follow the instructions correctly then?

different project but I tried that libfido2 link using the credential.py but it gives me an error which their docs don't really show much on configuring.
if it helps, the output I got from it
```
 python3 credential.py 
Use USB HID channel.
DEBUG: {'rk': True, 'up': True, 'alwaysUv': True, 'credMgmt': True, 'authnrCfg': True, 'clientPin': False, 'largeBlobs': True, 'pinUvAuthToken': True, 'setMinPINLength': True, 'makeCredUvNotRqd': True}
Authenticator supports User Verification
Traceback (most recent call last):
  File "/home/eitadmin/workspace/code/python-fido2/examples/credential.py", line 99, in <module>
    result = client.make_credential(create_options["publicKey"])
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eitadmin/miniconda3/envs/webauthn/lib/python3.11/site-packages/fido2/client.py", line 797, in make_credential
    return self._backend.do_make_credential(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eitadmin/miniconda3/envs/webauthn/lib/python3.11/site-packages/fido2/client.py", line 598, in do_make_credential
    pin_protocol, pin_token, pin_auth, internal_uv = self._get_auth_params(
                                                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eitadmin/miniconda3/envs/webauthn/lib/python3.11/site-packages/fido2/client.py", line 536, in _get_auth_params
    if self._should_use_uv(user_verification, mc) or permissions:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/eitadmin/miniconda3/envs/webauthn/lib/python3.11/site-packages/fido2/client.py", line 492, in _should_use_uv
    raise ClientError.ERR.CONFIGURATION_UNSUPPORTED(
fido2.client.ClientError: (<ERR.CONFIGURATION_UNSUPPORTED: 3>, 'User verification not configured/supported')
```
does any of this help identify some configuration required of the device?

Phizicks, 2023-12-30:
hm this one works and the led flashed and when clicked, it registered.
ok so it is working but wonder why it doesn't work on any test webauthn online service? not 100% compatible?

```
$ python3 verify_attestation.py
Use USB HID channel.

Touch your authenticator device now...

New credential created, attestation verified!
Yubico device AAGUID: db51083925a64424a33b9c800383b5c1
```

kaczmarczyck, 2024-01-03:
Ah right, the error message `User verification not configured/supported` above makes sense. You don't have a PIN set. Chrome usually asks you to do that on first usage. If you want to double check, you can also manually trigger it under "Manage security keys" in settings. Assuming that your particular problem doesn't prevent interaction with Chrome.

But it looks like your OpenSK works, and your OS can talk to it. You could try another website like https://webauthn.me/
Or maybe try a different browser, if you have one?

Phizicks, 2024-01-03:
ok, I managed to pull out an old windows 10 laptop and it works with the keys in chrome and firefox. so this issue is purely an Ubuntu 23.10 firefox problem.

yeah I tried that webauthn.me one as well. I just get the popup but never asked for pin setup or anything (unlike what happens on windows that I just checked).
[image]

ok. I guess this can be closed as it's really an ubuntu issue I need to find, unless someone knows the issue with solution.

kaczmarczyck, 2024-01-03:
It works on my Firefox `115.5.0esr` running on top of a different Debian derivate.

Thanks for the info and your debugging effort, it might help other Ubuntu users!

closed 2024-01-03 by kaczmarczyck

Phizicks, 2024-01-03:
ok, last update as this has been solved.
if you're using firefox and the device is not recognised or asking for pin etc, don't use the distro's packaged firefox but download from mozilla themselves. the issue seems to be potentially `snap` package related at least.

I downloaded a fresh firefox and it's prompting me now for a PIN and works *sigh*, after all that. :)

thanks for helping diagnosing and great work to the devs on the project.