google/OpenSK #678: How to delete an account you no longer use?

zvrh, 2024-03-01:
I have registered multiple test accounts on www.passkeys.io. How can I delete those accounts that are no longer in use on OpenSK?

[image]

## Steps to Reproduce the Problem

1. Register multiple different accounts on www.passkeys.io
1. Delete some of these accounts
1. Log in using passkey

## Specifications

- Version: 2.1
- Platform: Windows11 + Chrome

kaczmarczyck, 2024-03-01:
Since you use Chrome, you can go select "Sign-in data" in the Chrome settings:
chrome://settings/securityKeys

[image]

I assume these are discoverable credentials, meaning they are stored inside OpenSK. For non-resident keys, you'd have to either find a delete option inside your account at passkeys.io, meaning you wouldn't delete them on the device, but trust the website to delete their part. Or, as a last resort, you could "Reset your security key". Notice that you would delete everything, not just your credentials for that one website.

zvrh, 2024-03-11:
chrome://settings/securityKeys not found in my chrome , so I wrote a script that can be used to remove Opensk discoverable credentials:

```python
from fido2.hid import CtapHidDevice
from fido2.ctap2 import CredentialManagement, Ctap2, ClientPin
from fido2.ctap2.bio import BioEnrollment

class Ctap2Node:
    def __init__(self, connection):
        super().__init__()
        self.ctap = Ctap2(connection)
        self._info = self.ctap.info
        self.client_pin = ClientPin(self.ctap)
        self._auth_blocked = False
        self._token = None

    def unlock(self, pin):
        permissions = ClientPin.PERMISSION(0)
        if CredentialManagement.is_supported(self._info):
            permissions |= ClientPin.PERMISSION.CREDENTIAL_MGMT
        if BioEnrollment.is_supported(self._info):
            permissions |= ClientPin.PERMISSION.BIO_ENROLL
        self._token = self.client_pin.get_pin_token(pin, permissions)

    def cm(self):
        return CredentialManagement(
            self.ctap,
            pin_uv_protocol=self.client_pin.protocol,
            pin_uv_token=self._token,
        )

    def list_rp(self, cm: CredentialManagement):
        rps = cm.enumerate_rps()
        print(f'rps: {rps}')

def main(pin):
    devices = list(CtapHidDevice.list_devices())
    device = Ctap2Node(devices[0])
    device.unlock(pin)
    cm = device.cm()

    # list rp
    enumerate_rps = cm.enumerate_rps()
    print(f'\n\nenumerate_rps: {enumerate_rps}')
    for index, rp in enumerate(enumerate_rps):
        print(f'  {index}:{rp[CredentialManagement.RESULT.RP]["id"]}')
    index = input('select rp:').strip()
    select_rp = enumerate_rps[int(index)]

    # list cred
    enumerate_creds = cm.enumerate_creds(rp_id_hash=select_rp[CredentialManagement.RESULT.RP.RP_ID_HASH])
    print(f"\n\nenumerate_creds: {enumerate_creds}")
    for index, cred in enumerate(enumerate_creds):
        print(f'  {index}:{cred[CredentialManagement.RESULT.USER]["name"]}')
    index = input('select cred:').strip()
    select_cred = enumerate_creds[int(index)]

    # delete cred
    cm.delete_cred(select_cred[CredentialManagement.RESULT.CREDENTIAL_ID])

if __name__ == '__main__':
    main('xxxx') # set pin

```