google/OpenSK #685: `async` API for `Env`?

antonok-edm, 2024-05-06:
Hi! I'm working on [a project](https://kb2.antonok.com) using the [RTIC](https://rtic.rs/2/book/en/) framework on an STM32F4 microcontroller, and I'm interested in porting OpenSK to that architecture if at all possible.

RTIC leans heavily on Rust's `async` for resource sharing and task scheduling. Right now it looks like the current definition of traits required by `Env` exclusively use blocking function calls. After a first review for my use-case it looks like `HidConnection::send_and_maybe_recv`, `AttestationStore::{get,set}`, `Storage::{read_slice,write_slice,erase_page}`, and `UserPresence` would all need to access some kind of shared I/O peripherals. Under RTIC, non-blocking implementations of these traits would be awkward at best (and maybe highly unsafe at worst).

Before I go down this route, I wanted to check in with the maintainers - is this something you'd be willing to merge if a PR was provided? Any other thoughts?

kaczmarczyck, 2024-05-06:
Thank you for your interest. The current `Env` API has evolved with needs from different platforms. The API calls you mentioned are broadly in 2 categories.

### USB HID related

`HidConnection` and `UserPresence`.
Our current API is mostly taken from pre-library code and hasn't been redesigned yet. I could see us moving to an async API here.
We haven't touched it yet since we might want to be compatible with other transports like NFC in the future, and the new design should work for that.

### Storage related

`AttestationStore::{get,set}`, `Storage::{read_slice,write_slice,erase_page}`
First comment: This is work in progress, e.g., #679. I suggest you look at  the `develop` branch, not the current main branch `2.1` which is for people who want to build and use OpenSK more than it is for developers. Not too different yet, but more recent. `AttestationStore` moved into `Persist`.

Being async here is more problematic, since some of our security guarantees are tied to storage writes having persisted when execution continues. Can you help me understand why a blocking API doesn't work for you?

kaczmarczyck, 2024-05-08:
Question is how you'd design an `Env` that doesn't break all sync users and supports async use cases at the same time? To make sure I don't overpromise: I don't merge a PR that only moves our API to async.

So maybe it is indeed easier if you can use the sync OpenSK as is? Outside of user presence checking, our USB HID code is in the main loop, which is fully in your control. So the limitation that you can't run, e.g., display updates in parallel only applies when OpenSK processes a command.