google/OpenSK #702: Add Documentation for Running Tests in OpenSK Libraries

tklaback, 2024-09-10:
Hi all!

I am trying to run the tests located in /libraries/opensk/, but whenever I run cargo test from this directory, it fails with several errors. I'm likely missing some flags or feature toggles when running the tests and was wondering if there is any documentation on how to correctly run them.

One of the most common errors I encounter is:

error[E0433]: failed to resolve: use of undeclared type `Nat`
   --> <path_to_project_directory>/libraries/persistent_store/src/lib.rs:418:5
    |
418 |     Nat::try_from(x).unwrap()
    |     ^^^ use of undeclared type `Nat`

This is accompanied by warnings and additional errors related to unresolved imports (E0412, E0432, E0433).

For more information about these errors, I tried using rustc --explain E0412, but I believe this is due to some feature flags or target-specific configurations missing in my setup.

I've also tried running the tests with cargo test --all-features, but I'm still encountering the same issues. Could you please provide guidance on the correct steps, flags, or features required to run the tests successfully?

Maybe I am completely overlooking something but it would be great if this could be added to the documentation. Thank you!

kaczmarczyck, 2024-09-10:
Please have a look at the file [run_desktop_tests.sh](https://github.com/google/OpenSK/blob/2.1/run_desktop_tests.sh). For the OpenSK library, it has these commands (plus others that might be interesting for you):

```
echo "Running OpenSK library unit tests..."
cd libraries/opensk
cargo test --features std
cargo test --features std,config_command,with_ctap1
cargo test --all-features
```

I agree the section in our documentation that describes this is a bit hidden in [docs/customization.md](https://github.com/google/OpenSK/blob/2.1/docs/customization.md). Maybe we should have an extra README.md inside the library?

kaczmarczyck, 2024-09-10:
Oh, I just realized you did actually try

```
cargo test --all-features
```

I just tried it on a fresh clone and it works for me. Can you give me exact steps to reproduce and a full error log?

tklaback, 2024-09-10:
Yes, this fixed my issues, thank you for pointing this out. The reason `cargo test --all-features` didn't work was because I forked the repo and inadvertently created some breaking code. Thank you for your quick response.

closed 2024-09-10 by tklaback